MarlinFirmware/Marlin #22542: [BUG] Homing behavior changed in recent Marlin

marciot, 2021-08-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

This is a behavior which has been reported in some of our printers. I suspect it may be related to #22499, as it happened around the same time.

Previously, the Z would move down a little at the start of a homing move, then home twice.

Now the Z does not move down, and homes only once. This also causes it to not move at all if the endstop is triggered already at the start of a homing command. X and Y still double-tap the endstops while homing.

We should get the Z behaving how it used to, at least by having it double tap again.

### Bug Timeline

This happened sometime around 8/6

### Expected behavior

- The homing should start when homing to Z-min, even if the Z-max is triggered
- The homing should double-tap, just like X and Y.

### Actual behavior

- Homing to Z-min will not start if Z-max is triggered
- Homing on Z only causes a single tap.

### Version of Marlin Firmware

bugfix-2.0.x

### Printer model

Prototype printer

### Electronics

Archim 2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6955083/Configuration.zip)

thinkyhead, 2021-08-09:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

thinkyhead, 2021-08-13:
Note that it was never intentional that a Z max endstop could get triggered and interrupt motion when the Z homing direction is -1. I tried to add conditions to prevent Marlin reacting to an endstop unless it's specifically configured to check it. In response to your previous reports I tried simplifying the flags. I don't have any setup with endstops on both ends of an axis, so all I can think to do is to try reverting all the endstop flags to where they were around the time the 6-axis code, but that only takes us back to older unintentional behavior, and I will still want to fix them up in a more intentional way. I anticipate doing a lot more blind trial-and-error before I hit on exactly what you need.

I don't know how much coding and debugging you are comfortable with, but it would be nice to have some real collaboration on this issue because I am just taking stabs in the dark.

marciot, 2021-08-13:
> Note that it was never intentional that a Z max endstop could get triggered and interrupt motion when the Z homing direction is -1.

This problem has been solved with the changes you made. Thank you. There are two remaining behavioral changes:

1 - The homing does not initiate when the opposing endstop is triggered
2 - The double-tap does not seem to be working

> I don't know how much coding and debugging you are comfortable with, but it would be nice to have some real collaboration on this issue because I am just taking stabs in the dark.

It's pretty difficult for me to help out with the mechanical stuff when I do most of my development on a bare board that doesn't even have motors attached to it. Generally now a days my responsibilities have shifted to the FTDI Touch UI stuff and the wireless features in the SynDaver printers. I do have one printer I can use for testing, but that only runs the customized drunken-octopus firmware, so I can't really rely on features like "git bisect" because that discards all those local modifications and I can't get working firmware.

Would you happen to know which commit introduced the 6-axis code modifications? At this point, it may be best  for me to simply eyeball the code and see if anything jumps out at me.

marciot, 2021-08-13:
I am investigating. I will let you know if I find anything.

thinkyhead, 2021-08-14:
> The homing does not initiate when the opposing endstop is triggered

If everything was working according to intention, Marlin should be ignoring all endstops except the endstops at the homing end of the axis, and only when an axis is moving in the homing direction. And, Marlin should be completely unaware of extra endstops, only paying attention to endstops which are explicitly set up for homing. In other words, even when `USE_ZMAX_PLUG` is enabled Marlin is supposed to ignore that pin unless it has been explicitly set up as an endstop with `Z_HOME_DIR 1`.

> The double-tap does not seem to be working

Marlin should be ignoring the endstop when moving away from it. Maybe it is not taking account of the current direction of motion.

> Would you happen to know which commit introduced the 6-axis code modifications?

I recommend an editor or IDE that integrates with Git with a contextual menu item or command called "File History" or "Line History" which opens up a Git history browser of some kind. I use _Sublime Text 4_ along with _Sublime Merge_, and this produces a pretty good view of changes for a single file or a range of lines. The GitHub website isn't too bad for this purpose either, though it can be slow to show large diffs.

The main files to scrutinize are `endstops.cpp`, `endstops.h`, `Conditionals_post.h`, and maybe `G28.cpp`. Debug output may be added to `endstops.cpp` and `G28()` to trace what is going on at each stage of homing on your machine.

marciot, 2021-08-14:
> If everything was working according to intention, Marlin should be ignoring all endstops except the endstops at the homing end of the axis, and only when an axis is moving in the homing direction. [...]

I am not going to mince words on this. This behavior may be "intended", but it is wrong. Having a printer that occasionally crashes into an axis after a layer shift is fine for a home brew printer made out of plywood and rubber bands, but for any commercial printers out there, crashing an axis under any circumstance is unacceptable and there needs to be end stops on both sides of the axis and Marlin needs to respect those.

What I don't understand is that this correct behavior already exists in Marlin, but you are arguing to get rid of it.

I don't even suspect this is a hard thing to implement. Someone in Marlin the STEP and DIR lines are getting toggled. Maybe there needs to be a bitmask to allow motion in either the positive or negative direction of an axis and these will be updated based on the endstops. If any of those mask bits are cleared, the ISR will simply not step in that particular direction. I can't imagine this being more than a few lines of code.

> Marlin should be ignoring the endstop when moving away from it. Maybe it is not taking account of the current direction of motion.

I'm actually not sure whether this is wrong now. I need to consult with my coworker on monday as I am having trouble reproducing this.

I suspect after the last adjustments you made, things are in fact operating correctly, that is to say, the unintended way. But I should know more on Monday.

thinkyhead, 2021-08-14:
> there needs to be end stops on both sides of the axis and Marlin needs to respect those
> What I don't understand is that this correct behavior already exists in Marlin, but you are arguing to get rid of it.

What you don't understand is that the behavior is a side-effect and is not an explicit feature and I am arguing to make it an explicit feature, not to get rid of it.

DerAndere1, 2021-08-23:
To develop an explicit option to make Marlin alway stop at any triggered endstop, it might indeed help to have a look at the code before 6 axis support was added (relevant commits in chronological order: https://github.com/MarlinFirmware/Marlin/commit/02f904dbf9df999264099a26c563b07b1d163df4, https://github.com/MarlinFirmware/Marlin/commit/92dea8e6ccd26950eed817fce6f574fcfe866489, https://github.com/MarlinFirmware/Marlin/commit/dd4990252e891cdfe56cb7d6e3bbe1e6289be649, https://github.com/MarlinFirmware/Marlin/pull/19112/commits).

badbod, 2021-08-24:
I am using Ender 3 V2 and I can confirm the single 'tap' Z homing issue with BLTouch.  From the center of the bed, if I auto home or  just home Z,  it only single probes instead of the fast probe-slow probe 'double tap' I am used to. 
Setting is #define MULTIPLE_PROBING 2

I use
     #define SENSORLESS_HOMING
for X/Y, In case that affects anything?

Only observed in recent bugfix-2.0.x. 

As I do not have any physical end stops I can not confirm the issue of failing to home while triggered max end stop

badbod, 2021-09-03:
which was the last version that homing worked properly? I would like my double tap back on BLTouch, no matter what I set for homing options and BLTouch options, homing only taps once. Align-Z , UBL mesh ABL etc all double tap (or whatever I set in multi/additional probing) just G28 , or G34/G29 initial home is affected.

thinkyhead, 2021-09-11:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

thinkyhead, 2021-09-11:
For those with Z homing bump issues, what does your `HOMING_BUMP_MM` setting look like?

mydevpeeps, 2021-09-22:
I am using the hotfix from July, I have a bltouch on the old z endstop, it double taps for home z. I am not using sensorless homing anywhere. I'm on a creality v2.2 silent board. anything else you want to know or have me try?

closed 2021-10-17 by thinkyhead