MarlinFirmware/Marlin #22544: UBL will not put the mesh in selected slot.

BreeOge, 2021-08-09:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When using the Display on the 3D printer, and when you select what UBL slot you wish to go into, it shows the correct one.  However if you watch the console, it saves it to slot 0 no matter what.

### Bug Timeline

New

### Expected behavior

When selecting the UBL slot and running the UBL mesh wizard, you expect it to save in the slot you selected.

### Actual behavior

When selecting the UBL slot and running the UBL mesh wizard, it saves in slot 0 no matter what slot you select.

### Steps to Reproduce

Use the UBL mesh Wizard.
Select Slot you want active.
Run Wizard
Watch Console for saved slot.

### Version of Marlin Firmware

Marlin 2.0 Bugfix Latest

### Printer model

Geeetech A20T

### Electronics

Stock

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

Vertabreak, 2021-08-09:
will look it over but the default slot that is loaded is always 0 you have to manually load other slots.

Vertabreak, 2021-08-10:
so i just ran a test with M1004 S10 and it does save to slot 10 as intended the issues seems to be with the menu item it self not passing the selected slot off to the gcode being executed.

Vertabreak, 2021-08-10:
testing a few tweaks to see if they correct the issue.

tpruvot, 2021-08-12:
Cant reproduce the problem, with lcd menu... (CLASSIC_UI) selected slot 1 and then selected store mesh

```
23:28:15.433 > Unified Bed Leveling System v1.01 inactive
23:28:15.483 > Mesh loaded from slot 0
23:28:15.483 > Mesh 0 loaded from storage.
23:28:20.120 > echo:SD card ok
23:28:38.581 > echo:G29S1
23:28:38.581 > M117 Grille 1 enreg.
23:28:38.660 >
23:28:38.709 > Mesh saved in slot 1
23:28:38.710 > Done.
```

Vertabreak, 2021-08-13:
strange i am able to reproduce on my A20T with full gfx display.
to be clear this is a issue with the Mesh Wizard Menu entry not a UBL issue or a issue with the Gcode back end of the mesh wizard.

thinkyhead, 2021-08-13:
Strange. The menu has its own static `ubl_storage_slot` value that it passes to `M1004` or `G29 S`. Does adding extra logging to `_lcd_ubl_load_save_cmd` and/or `_lcd_ubl_mesh_wizard` show that variable getting stepped on? Does `M111 S3` show anything unusual going into the queue? Is `char ubl_lcd_gcode[30]` inside `_lcd_ubl_mesh_wizard` too small a buffer?

Vertabreak, 2021-08-13:
i tested with a buffer of 64 it did not fix the issue more analysis is needed.

thinkyhead, 2021-09-11:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Vertabreak, 2021-09-11:
the menu option always saves to slot0 still and the M1004 S# still functions correctly unsure why the slot# is not getting passed from the menu selection.
i ran the test on today's bugfix on my A10.

thinkyhead, 2021-09-13:
I wonder why. Would it help if someone helped you to add serial logging to the code at the points where it is producing the wrong slot # so that we can see where the problem originates?

Vertabreak, 2021-09-13:
the issue only presents from the menu i have tried a number of tweaks to code looking for new results and so far have found none. it seems like the 3rd variable is just ignored always in the menu defaulting to slot 0. i have exhausted my ideas and am open to suggestions, probably some minor detail everyone is overlooking or something.

sjasonsmith, 2021-11-25:
@Vertabreak can you still reproduce this problem? It seems fine when I test it in the simulator. It could possibly be a problem only on some architectures.

closed 2022-01-10 by thinkyhead