MarlinFirmware/Marlin #22545: [BUG] Default FAN_PIN stuck in ON state Even if reassigned

jatmn, 2021-08-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I was working on remapping the pins for the fan and heaters on my custom build using an Archim2 and noticed I was not able to get my hotend cooling fan to work with min temp to auto shut off.  I noticed when I reassigned to a different pin that was NOT FAN_PIN's Default assinged pin it worked just fine..

Archim2 features a LED indicator next to each fan and heater header. 
When I reassigned the FAN_PIN to 7 which was originally HEATER_1_PIN, I noticed the original header which is designated as pin 4 was still active. The led indicator for Pin 4 (the default FAN_PIN noted as FAN 0 on the schematics from ultimachine) is still on. Even if I redefine Pin 4 to some other name the Pin is still "on"

I confirmed this in latest Bugfix as of the time reporting this issue.. as well as bugfix downloaded yesterday. 
I cant confirm if this is a issue in the latest release build as there is a bug in that build that prevents compiling for Archim2

### Bug Timeline

New

### Expected behavior

Pin 4 (FAN 0) would not be active if undefined or its purpose was reassigned.

### Actual behavior

Pin 4 stays active even if undefined

### Steps to Reproduce

open sam/pins_ARCHIM2.h 

reassign FAN_PIN from 4 to 7 (or any other pin)
[test.zip](https://github.com/MarlinFirmware/Marlin/files/6959064/test.zip)

### Version of Marlin Firmware

Latest Bugfix as of this report

### Electronics

Archim 2

thinkyhead, 2021-08-10:
You may be able to enable `DIRECT_PIN_CONTROL` and use `M42` to turn the pin on and off directly, by first forcing `FAN_PIN` to -1. That could tell us if something else is messing with the pin. You may also use `PINS_DEBUGGING` with `M43` to see if any pins are overlapping.

jatmn, 2021-08-10:
Here are the results of that test..
[M43 before setting FAN_PIN -1.txt](https://github.com/MarlinFirmware/Marlin/files/6964992/M43.before.setting.FAN_PIN.-1.txt)
[M43 with setting FAN_PIN -1.txt](https://github.com/MarlinFirmware/Marlin/files/6964993/M43.with.setting.FAN_PIN.-1.txt)

The pin however *IS* still active and powering whatever is plugged into it.

thinkyhead, 2021-08-13:
With `FAN_PIN` defined as -1, you can try `M42 P7 S0` to turn the pin off directly. If it has no effect on the pin's voltage then it may just be stuck on at the hardware level, indicating a short.

closed 2021-09-20 by thinkyhead