MarlinFirmware/Marlin #22546: [BUG] CR-10 v3 Z-stop stays triggered. [Help Wanted]

prfpinguin, 2021-08-10:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

With my Marlin version for the CR-10 V3 upgraded with a Hemera and Super-Volcano the Z-stop stays triggered either inverted or normal (M119). 
I have replaced both cable and limit switch and have switched them with the Y-stop to make sure the hardware is not faulty. 

The X and Y stop work as expected.

I expected the BL-touch sensor settings to interfere with the z-stop settings somewhere, but I fail to see where.

### Bug Timeline

New

### Expected behavior

Z-stop state to change from open to triggered when Pressed.

### Actual behavior

Z-stop stays triggered no matter the trigger state.

### Version of Marlin Firmware

Marlin 2.0.9_DW7.4

### Printer model

Creality CR-10 V3

### Add-ons

Hemera, Super Volcano

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/6961539/Marlin.zip)

thinkyhead, 2021-08-10:
Please test to see if the same problem exists in Marlin 2.0.9.1 from [here](https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.0.9.1.zip). You may be seeing a regressing from recent endstop changes.

sjasonsmith, 2021-11-25:
Closing due to lack of activity and response from submitter.

closed 2021-11-25 by sjasonsmith