MarlinFirmware/Marlin #22554: TFT_GENERIC on SKR 2 REV A (error: 'AUTO' does not name a type / error: 'ILI9341' does not name a type)

joaocqmiranda, 2021-08-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Initially this error appeared: 'TFT_A0_PIN' / 'TFT_CS_PIN' was not declared in this scope; did you mean 'TFT_SPI'?

Then add this code to the pins_BTT_SKR_v2_0_common.h file:
[https://github.com/MarlinFirmware/Marlin/issues/19933](url)

   ```
 ////////////////////////////////////////////
  #elif HAS_SPI_TFT                               // Config for Classic UI (emulated DOGM) and Color UI
    #define TFT_CS_PIN                     EXP1_04_PIN
    #define TFT_A0_PIN                     EXP1_03_PIN
    #define TFT_DC_PIN                     EXP1_03_PIN
    #define TFT_MISO_PIN                   EXP2_10_PIN
    #define TFT_BACKLIGHT_PIN              EXP1_08_PIN
    #define TFT_RESET_PIN                  EXP1_07_PIN

    #define TFT_SCK_PIN                    EXP2_09_PIN
    #define TFT_MOSI_PIN                   EXP2_10_PIN

    #define LCD_USE_DMA_SPI

    #define TOUCH_INT_PIN                  EXP1_05_PIN
    #define TOUCH_CS_PIN                   EXP1_06_PIN
    #define TOUCH_BUTTONS_HW_SPI
    #define TOUCH_BUTTONS_HW_SPI_DEVICE        2

    // Disable any LCD related PINs config
    #define LCD_PINS_ENABLE                -1
    #define LCD_PINS_RS                    -1

    #ifndef TFT_BUFFER_SIZE
      #define TFT_BUFFER_SIZE               1200
    #endif
    #ifndef TFT_QUEUE_SIZE
      #define TFT_QUEUE_SIZE                6144
    #endif

    #define BTN_EN1                        EXP2_08_PIN
    #define BTN_EN2                        EXP2_06_PIN
    /////////////////////////////////////////////
```

And now this error appears: 'AUTO' does not name a type

### Version of Marlin Firmware

2.0.9

### Printer model

Anet A6

### Electronics

 SKR 2 REV A

### Add-ons

BL touch, Run out filament

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Marlin-2.0.9-SKR.zip](https://github.com/MarlinFirmware/Marlin/files/6970422/Marlin-2.0.9-SKR.zip)

thinkyhead, 2021-08-12:
Please provide configurations that apply to `bugfix-2.0.x` so that we can build against the current code and see the error that you are seeing. I have no idea what we are supposed to do with a folder containing the entire Marlin source distribution.

joaocqmiranda, 2021-08-12:
On that folder are also the files that I configured. But i attached now only the files that i edited. Is this what you wanted?
[Configuration_TFTG_SKR2.zip](https://github.com/MarlinFirmware/Marlin/files/6974899/Configuration_TFTG_SKR2.zip)

ellensp, 2021-08-12:
Cannot reproduce on bugfix.

Config files updated for bugfix
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6976510/Configuration.zip)

The only change I had to make was #define TEMP_SENSOR_0 1 as 69 is not a marlin provided table.

joaocqmiranda, 2021-08-12:
> Cannot reproduce on bugfix.
> 
> The Config files updated for bugfix, just drop in and compile.
> [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6976510/Configuration.zip)
> 
> The only change I had to make to your configurations files was #define TEMP_SENSOR_0 1 as 69 is not a marlin provided table.
> 
> 
> 
Thanks. Yes i forget to send that file that I created. I will try.

joaocqmiranda, 2021-08-15:
Yes I managed to compile, but when I flash the printer the fans must be on when the 3d printer is on. as it did in the previous configuration.

I have another question (image). How do I connect this ldc to the motherboard? Do I need an adapter?
I connected the pins in common but there were still pins that I couldn't connect to the LCD

[image]

ellensp, 2021-08-15:
you cant. It has the wrong type of interface  
Marlin expects a SPI interface, that is a 8 bit parallel interface
Also that board is designed for 5v signals, see all the chips, those convert the 5v down to 3.3v needed by the lcd.

closed 2021-08-15 by ellensp