MarlinFirmware/Marlin #22555: [FR] G35 semi automatic Assisted Tramming

javicarrera, 2021-08-11:
### Describe the feature you want

I was thinking that would be great to improve the feature G35 Assisted Tramming as follows:
If you have any kind of probe the G35 code could go through all corners reading the heights, take the closest corner from probe to bed as a reference and go to each of the rest 3 corners. In each corner would pause so we can lift (always lift) the corner until probe triggers and then automatically continues to next corner. 

In this way we would get always perfect flat surface, without any paper trick.
If you don't have a probe would stay as it is.
I called it "semi automatic" because you still have to turn the wheels.

What do you think?

thinkyhead, 2021-08-11:
Have you tried the `ASSISTED_TRAMMING_WIZARD` or `LEVEL_CORNERS_USE_PROBE`? Presumably they both work in a way similar to this.

javicarrera, 2021-08-12:
You are right! Didn't notice `LEVEL_CORNERS_USE_PROBE` option I will try later but I think that's it. Thanks again.