MarlinFirmware/Marlin #22556: [BUG] (skr 2 rev b, stock ender 3 display not displaying anything)

CyberClu, 2021-08-11:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am new to 3d printing and tried to upgrade my ender 3 mainboard to the BTT SKR 2 and I got a rev b board. The issue is that when I turn on the printer the display lights up, but does not display anything. I can connect to the board via USB and can control it just fine and when I flash the firmware it flashes correctly because there is a cur file on the SD card after I flash. I have plugged the display cable into EXP 1 on the SKR 2 mainboard and into EXP 3 on the display. In the marlin firmware I uncommented #define CR10_STOCKDISPLAY and defined the following delays in Configuration.h #define ST7920_DELAY_1 DELAY_NS(100), #define ST7920_DELAY_2 DELAY_NS(900), #define ST7920_DELAY_3 DELAY_NS(100).  The editor that I am using is VS Code and I am using marlin auto builder extension and the Platform IO IDE extension. the version of marlin is BTT version but I just changed the display to the cr10 stock. I have tried flipping the display connector to the mainboard and buying a new display. I have also tried configuring marlin from a default version of marlin downloaded directly from the GitHub.

### Bug Timeline

about a week and a half ago when I first tried my new mainboard

### Expected behavior

I expect the Marlin boot screen to show up and then to be able to see the menu and navigate through it

### Actual behavior

The stock ender 3 lcd lights up but does not show anything.

### Steps to Reproduce

i am not sure exactly if this will reproduce the problem but what I did is;

1.  Open up the [BTT skr 2 preconfigured marlin](https://github.com/bigtreetech/SKR-2/tree/master/Firmware/Marlin-bugfix-2.0.x) 
2.  make sure it is setup for tmc 2209
3.  make sure the only display that is uncommented is the #define CR10_STOCKDISPLAY
4.  set delays to the following: 
        #define ST7920_DELAY_1 DELAY_NS(100)
        #define ST7920_DELAY_2 DELAY_NS(900)
        #define ST7920_DELAY_3 DELAY_NS(100)
5.  compile using marlin auto builder 
6.  plug in the ender 3 stock display into the exp 1 on the board and then into exp 3 on the display
7.  insert sd card and turn on printer to flash firmware

### Version of Marlin Firmware

BTT skr 2 preconfigured

### Printer model

ender 3

### Electronics

skr 2, TMC 2209 stepper motor drivers, and base ender 3

### Add-ons

just the new main board

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[Marlin-bugfix-2.0.x045.zip](https://github.com/MarlinFirmware/Marlin/files/6971506/Marlin-bugfix-2.0.x045.zip)

this is my marlin configuration I just added the 045 so I could distinguish it from other marlin folders

ellensp, 2021-08-12:
provided configs has CONFIGURATION_H_VERSION 020008  this is old 
current pins files for this board has different lcd timings to current code

#if HAS_MARLINUI_U8GLIB
  #ifndef BOARD_ST7920_DELAY_1
    #define BOARD_ST7920_DELAY_1   DELAY_NS(120)
  #endif
  #ifndef BOARD_ST7920_DELAY_2
    #define BOARD_ST7920_DELAY_2    DELAY_NS(80)
  #endif
  #ifndef BOARD_ST7920_DELAY_3
    #define BOARD_ST7920_DELAY_3   DELAY_NS(580)
  #endif
#endif

CyberClu, 2021-08-12:
> provided configs has CONFIGURATION_H_VERSION 020008 this is old
> current pins files for this board has different lcd timings to current code
>
> #if HAS_MARLINUI_U8GLIB
> #ifndef BOARD_ST7920_DELAY_1 [...]

If I add that to my current Configuration.h would it work or should I go to a new version? because I tried setting up the newest bugfix and the problem continued.

thinkyhead, 2021-08-13:
Yes, you can edit them in `Configuration.h` or you can comment them out for the defaults from the pins file.
```cpp
#define ST7920_DELAY_1 DELAY_NS(100)
#define ST7920_DELAY_2 DELAY_NS(900)
#define ST7920_DELAY_3 DELAY_NS(100)
```
🤷🏻  Try a variety of delays and maybe one of them will work.

CyberClu, 2021-08-13:
> Yes, you can edit them in `Configuration.h` or you can comment them out for the defaults from the pins file.
> 
> ```c++
> #define ST7920_DELAY_1 DELAY_NS(100)
> #define ST7920_DELAY_2 DELAY_NS(900)
> #define ST7920_DELAY_3 DELAY_NS(100)
> ```
> 
> 🤷🏻 Try a variety of delays and maybe one of them will work.

Ok Thanks I'm on vacation right now I will be back in few days and I will try some delays and see which one works I think my problem might have been that I hadn't tried very many delays. So I will try that and see how it works.

CyberClu, 2021-08-15:
I am now able to test different timings, but I was wonder if there is ant type of ratio between the 3 delays?

ellensp, 2021-08-15:
The important bit of code is 
```
#define ST7920_SND_BIT do{ \
  WRITE(ST7920_CLK_PIN, LOW);             ST7920_DELAY_1; \
  WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
  WRITE(ST7920_CLK_PIN, HIGH);            ST7920_DELAY_3; \
  val <<= 1; }while(0)
```

ST7920_DELAY_1 is delay from clock low to setting data bit 
ST7920_DELAY_2 is delay from data bit to clock high
ST7920_DELAY_3 is delay between bits

These are the common values for stm32f4 based boards

ST7920_DELAY_1   96 - 125
ST7920_DELAY_2   48 - 90
ST7920_DELAY_3  500 - 715

CyberClu, 2021-08-15:
how accurate do the numbers have to be to start showing something at all on the screen?

CyberClu, 2021-08-18:
> The important bit of code is
> 
> ```c++
> #define ST7920_SND_BIT do{ \
>   WRITE(ST7920_CLK_PIN, LOW);             ST7920_DELAY_1; \
>   WRITE(ST7920_DAT_PIN, ST7920_DAT(val)); ST7920_DELAY_2; \
>   WRITE(ST7920_CLK_PIN, HIGH);            ST7920_DELAY_3; \
>   val <<= 1; }while(0)
> ```
> 
> ST7920_DELAY_1 is delay from clock low to setting data bit
> ST7920_DELAY_2 is delay from data bit to clock high
> ST7920_DELAY_3 is delay between bits
> 
> These are the common ranges for stm32f4 based boards
> 
> ST7920_DELAY_1 96 - 125
> ST7920_DELAY_2 48 - 90
> ST7920_DELAY_3 580 - 715

around how accurate do the numbers have to be to start showing something at all on the screen?

thinkyhead, 2021-08-19:
> how accurate do the numbers have to be

Start with very large values, guaranteed to produce a display (but wasting more processor time), and then reduce them until you find the lowest values that work.

CyberClu, 2021-08-19:
> > how accurate do the numbers have to be
> 
> Start with very large values, guaranteed to produce a display (but wasting more processor time), and then reduce them until you find the lowest values that work.

I set the delays to 2000 each and flashed the firmware and the display stayed blank. Is this normal?

CyberClu, 2021-08-19:
Never mind I got It to work I decided to try the other display I had ordered again to see if it would work with the higher timings and it turns out that that fixed it I had originally thought that I tried the new display with the new timings but apparently I hadn't. Thanks for all the help and support I apparently didn't take the time to think through everything. Thanks again

closed 2021-08-19 by CyberClu