MarlinFirmware/Marlin #22561: Z axis not homing at the center, and the TFT level screen locations are opposite

slf495, 2021-08-12:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

   upgraded to skr 2, and for a time homing was way off, and got it sorted out, but i can't get the Z axis to home at the center of the bed, it's off, and when it goes to ABL it goes to the same wrong spot, and will then move to the front of the bed but won't stop
    also, when trying to use the TFT touch screen to manually move the nozzle,  instead of moving to the front left as selected, it moves to the exact opposite area

### Bug Timeline

new, with the new board

### Expected behavior

the z axis should be homing at the center of the bed, and when it goes to ABL, it should move to the front left, and probe to the right. Also, with the touch level selections, it should move to the appropriate locations

### Actual behavior

it's not moving to the center of the bed when homing z. and when selecting the manual location on the touch screen it's moving to the opposite side

### Version of Marlin Firmware

the most recent bugfix 2.0

### Printer model

ender 5 pro

### Electronics

tft touch screen, skr 2 board (new)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/6976945/configs.zip)

thisiskeithb, 2021-08-12:
This is a configuration issue. First example is:

```cpp

thisiskeithb, 2021-08-12:
This is a configuration issue. First example is:

```cpp
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
```

The Ender-5 & Pro use X max & Y max plugs. Which would explain why everything is 180 out.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-12 by thisiskeithb

slf495, 2021-08-12:
Ok, so maybe (more than maybe) I’m dumb, but the skr 2 board has it setup with “X-Stop” on the board, so don’t you have to use the min instead of the max?

slf495, 2021-08-12:
Ok so I changed the min to max, and changed the value from -1 to 1 on the home, and I am back to it only moving a few mm’s and then stopping when homing