MarlinFirmware/Marlin #22563: [BUG] I can not position with DELTA type

gear2nd-droid, 2021-08-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When DELTA positioning is performed, the target coordinates are not allowed to be able to position and can not be positioned.
When positioning in the XY plane, coordinates are changed to X = 0, Y = 0, Z = 0 (origin). In addition, since it is moving to the origin, all subsequent positioning commands are going to the origin and will not move.
In the z-axis direction, it operates properly. However, when the movement of the XY plane is sandwiched at one time, Z = 0 is.

### Expected behavior

After returning home positioning, positioning in the XY plane.

### Actual behavior

After completion of the home position return, positioning is performed in the XY plane, X = 0, Y = 0, Z = 0 is positioned once. Also, the positioning in the following XY plane is all X = 0, Y = 0, Z = 0.

### Steps to Reproduce

1.First Step : homing
2.SecondStep : positioning on XY plane

### Version of Marlin Firmware

2.0.9.1

### Electronics

FLYF407ZG

### Host Software

Pronterface

thinkyhead, 2021-08-14:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where positioning works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 days in the past, it would take no more than 8 tests to find the exact day when it broke.

gear2nd-droid, 2021-08-14:
How do I get older versions?

thinkyhead, 2021-08-14:
From this page you can browse Marlin `bugfix-2.0.x` commits and download the firmware from any point in time:

https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x

gear2nd-droid, 2021-08-16:
I was able to go back to version 2.0.6 (around September 2020), which I do not know in the relationship of the board you are using.
Symptoms have occurred. Is my setting funny?

Configuration.h
#define DELTA
#if ENABLED(DELTA)
  #define DELTA_SEGMENTS_PER_SECOND 200
  #if ENABLED(DELTA_AUTO_CALIBRATION)
    #define DELTA_CALIBRATION_DEFAULT_POINTS 4
  #endif
  #if EITHER(DELTA_AUTO_CALIBRATION, DELTA_CALIBRATION_MENU)
    #define PROBE_MANUALLY_STEP 0.05
  #endif
  #define DELTA_PRINTABLE_RADIUS 90.0
  #define DELTA_DIAGONAL_ROD 215.0
  #define DELTA_HEIGHT 250.00
  #define DELTA_ENDSTOP_ADJ { 0.0, 0.0, 0.0 }
  #define DELTA_RADIUS 105.2
  #define DELTA_TOWER_ANGLE_TRIM { 0.0, 0.0, 0.0 }
#endif
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG
#define CLASSIC_JERK
#define X_HOME_DIR 1
#define Y_HOME_DIR 1
#define Z_HOME_DIR 1

thinkyhead, 2021-08-19:
Maybe one of our contributors is more familiar with deltabots and has some ideas. Visit the Marlin Discord server and ask around. Join link: https://discord.gg/n5NJ59y

gear2nd-droid, 2021-08-19:
thank you.  I can't write to discord.

sjasonsmith, 2021-11-25:
@gear2nd-droid are you still interested in this? I have used my delta recently and it seemed ok. I don't want to spend time investigating this if you have already moved on.

gear2nd-droid, 2021-11-25:
Now it moves.  Thank you very much.

closed 2021-11-25 by ellensp