MarlinFirmware/Marlin #22568: [BUG] M43 on LPC1768 reporting analog pins as <unused/unknown>

chschu, 2021-08-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Pin debug (M43) output on LPC1768 does not report the usages of analog pins correctly.

### Bug Timeline

Probably introduced with #15722

### Expected behavior

The output of M43 should contain analog pin usages like `TEMP_0_PIN`, as it does for other target platforms. The ADC channel number should be reported correctly.

### Actual behavior

Analog pins that are actually used (e.g. as `TEMP_0_PIN`) are reported as unused, and with `(A<pin>)` instead of (A<channel>)`:

```
PIN: 0.24 (A24)  <unused/unknown>   Analog in =   973   Input  = 0
```

### Steps to Reproduce

- Build for LPC1789 (or LPC1769) with `PINS_DEBUGGING` defined and `TEMP_0_PIN` defined as `P0_24_A1`.
- Flash firmware
- Send M43
- Check output for analog pins

### Version of Marlin Firmware

2.0.9.1

### Printer model

Anycubic Mega X

### Electronics

BTT SKR 1.4 Turbo

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Additional information & file uploads

The issue seems to be related to b9116d4050c2ac0529fb7f63815ea47a99904a80. In this commit, the pin definitions for analog pins on LPC1768/9 boards have been changed from ADC channel numbers (0-7) to the actual pin numbers (2-3, 23-26, 62-63).

This causes `ANALOG_OK` in `pinsDebug_list.h` to fail for two reasons:
- The Arduino library for LPC1768 does not define `NUM_ANALOG_INPUTS`, but declares it as a `constexpr`. This means that `NUM_ANALOG_INPUTS` is undefined for the preprocessor, and `(PN) < NUM_ANALOG_INPUTS` is never `true`.
- Even if `NUM_ANALOG_INPUTS` had the correct value (`8`), the comparison would still fail for 23-26 and 62-63.

Side note: Many pin definitions (especially the "mega" ones) use ADC channel numbers, while STM32 and LPC1768/9 use the actual pins.

chschu, 2021-08-19:
Regarding the last paragraph of the bug description: According to [the docs](https://marlinfw.org/docs/hardware/boards.html#analog-pins), analog pins should be assigned using their index (that's what I called "ADC channel number").

ellensp, 2021-08-19:
That page is obsolete and only refers to 8 bit controllers.

thisiskeithb, 2021-08-23:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22611

closed 2021-08-23 by thisiskeithb