MarlinFirmware/Marlin #22569: [BUG] (Z safe homing acceleration)

Malderin, 2021-08-15:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When performing an auto home with the Z safe homing option enabled, there is a large acceleration along the X and Y axes.
For manual movements, the accelerations are normal, as defined by the configuration file.

Board: MKS Robin nano v.2 whih MKSUI interface

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/6987758/Configuration.zip)

Bug video: https://disk.yandex.ru/d/iXljObCX9eEF7g

### Version of Marlin Firmware

bugfix 2.0

### Electronics

MKS Robin nano v.2

mayy42, 2021-08-15:
I have the same problem with the 2.0.9.1 from Aug 10, 2021.
My printer can not handle the acceleration and the y-axis stops after half of the way and makes strange noises.

roel8032, 2021-08-15:
Please try M501 before G28 (G29).

Malderin, 2021-08-15:
> Please try M501 before G28 (G29).

without changes

ManuelMcLure, 2021-08-15:
I’m seeing this as well with recent bugfix-2.0.x (around a couple days old).

ManuelMcLure, 2021-08-15:
My gut feeling is that it is due to 9130f58f3f553584278ec716c617005b9e03cb49

thisiskeithb, 2021-08-18:
https://github.com/MarlinFirmware/Marlin/pull/22574 was merged. Closing.

closed 2021-08-18 by thisiskeithb