MarlinFirmware/Marlin #22573: [BUG] Build fails with MKS SGEN L and MKS MINI 12864 Panel

nicknugget, 2021-08-16:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to build marlin for a Tevo Tarantula Pro RS. The printer has a 32bit MKS SGEN L mainboard and uses a MKS Minipanel (https://reprap.org/wiki/MKS_MINI_12864)
 
I'm using VS Code Version 1.59.0

The build process fails with following error:

```
In file included from Marlin/src/lcd/dogm/marlinui_DOGM.cpp:42:
Marlin/src/lcd/dogm/marlinui_DOGM.cpp: In static member function 'static void MarlinUI::init_lcd()':
Marlin/src/lcd/dogm/marlinui_DOGM.h:225:23: error: 'DOGLCD_CS' was not declared in this scope; did you mean 'DOGLCD_SCK'?
  225 |     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
      |                       ^~~~~~~~~
Marlin/src/lcd/dogm/marlinui_DOGM.cpp:274:14: note: in expansion of macro 'U8G_PARAM'
  274 |     u8g.init(U8G_PARAM);
      |              ^~~~~~~~~
Marlin/src/lcd/dogm/marlinui_DOGM.h:225:34: error: 'DOGLCD_A0' was not declared in this scope; did you mean 'DOGLCD_SCK'?
  225 |     #define U8G_PARAM DOGLCD_CS, DOGLCD_A0                      // HW-SPI
      |                                  ^~~~~~~~~
Marlin/src/lcd/dogm/marlinui_DOGM.cpp:274:14: note: in expansion of macro 'U8G_PARAM'
  274 |     u8g.init(U8G_PARAM);
      |              ^~~~~~~~~
*** [.pio/build/LPC1768/src/src/lcd/dogm/marlinui_DOGM.cpp.o] Error 1
```

### Bug Timeline

New bug

### Expected behavior

I expect a successful firmware build

### Actual behavior

Build process fails with errors related to the selected display

### Steps to Reproduce

Use attached config
[Marlin_Config.zip](https://github.com/MarlinFirmware/Marlin/files/6995220/Marlin_Config.zip)

Start build process in VSCode with PlatformIO

### Version of Marlin Firmware

2.0.9.1

### Printer model

Tevo Tarantula Pro RS 

### Electronics

Stock electronis, MKS SGEN L mainboard

### Your Slicer

Cura

thisiskeithb, 2021-08-16:
Enable `MKS_MINI_12864` and disable `MINIPANEL`.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-16 by thisiskeithb

nicknugget, 2021-08-16:
Thanks for the feedback and sorry for the trouble. After submitting the report I found the problem (need to comment in "#define SDSUPPORT" AND use "MKS_MINI_12864")