MarlinFirmware/Marlin #22575: [FR] 7 Axies and sensor feedback

i-make-robots, 2021-08-16:
### Is your feature request related to a problem? Please describe.

I'm getting old and can't support my firmware any more.  If I can use Marlin instead of my crappy Marlin variant then a lot of stress goes away.

### Are you looking for hardware support?

reprap discount rumba controller (megaatmega2560)
BIGTREE_SKR_PRO 1.2
TMC2130
AS5147 angle sensor

### Describe the feature you want

All my robots are stepper driven systems in various form-factors: polargraphs, arms, deltas, stewart platforms - that means 2-6 stepper motors and >=1 servos.  

### Additional context

I can do all the IK off-board and generate gcode with absolute positions, so ik and fk do not need to happen on-board.  more cpu/step!   All that's missing here from Marlin is support for axies 5&6.  With extruder that would make it at least 7 axies.
I'm good with any axis labels.  I've been using XYZUVW.

I would also like to add support for AS5147 sensors but will save that for a future FR.

thisiskeithb, 2021-08-16:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22488

thisiskeithb, 2021-08-16:
> I would also like to add support for AS5147 sensors but will save that for a future FR.

Yes, please!

closed 2021-08-16 by thisiskeithb

DerAndere1, 2021-08-20:
Marlin has support for 6 axes plus extruders (= 7) already, so XYZUVWE is already possible. Even more axes are reportedly working now with my fork (https://github.com/DerAndere1/Marlin/tree/9axis_pull) . It was tested once with BED_LEVELING_3POINT . Feel free to test and report issues at https://github.com/DerAndere1/Marlin/issues . To fix other bed leveling methods, my 9axis_pull branch has to be rebased onto current Marlin bugfix-2.0.x .