MarlinFirmware/Marlin #22576: [FR] Manual Bed Levelling Should require preheat for bed and nozzle

Barrnet, 2021-08-16:
### Describe the feature you want

Like for the Automatic Bed Levelling, the Mesh Bed Levelling should preheat bed and nozzle (maybe at only 160 Celsius for not having oozing problem) if PREHEAT_BEFORE_LEVELING is defined.

thisiskeithb, 2021-08-16:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/22549

closed 2021-08-16 by thisiskeithb

Barrnet, 2021-08-17:
Is not a duplicate. I was talking about Mesh Bed Levelling, not Bed Tramming.

sarvenn, 2021-08-28:
Agree. I solved this setting a preheat selection 150C / 60C.