MarlinFirmware/Marlin #22580: [FR] Add support for CNC-style jogging

dancollins, 2021-08-17:
### Is your feature request related to a problem? Please describe.

Trying to jog a CNC machine running Marlin firmware will often have a bit of latency, as sometimes you've added too many moves to the planner by the time the machine is where you'd like it.

Performing a quick stop, with M410, or waiting for motion to finish with M400 will (probably, I haven't tested this) result in either stopping too fast (and jerking out of position) or not stopping soon enough.

### Describe the feature you want

I've built an MPCNC, and I use Marlin for my firmware (on a BTT SKR 1.4 Turbo, LPC1769), mostly because I'm familiar with it and it works really well.

When using the machine, a standard operation is to move the head to a point on the workpiece and set XYZ to zero (G92). On other machines, like the Hass, there's a wheel that lets you jog nice and quickly. CNCjs supports this - but really it's all just gcode and I think only GRBL has a 'stop jogging now' feature.

What I'd like to do is implement another gcode for Marlin that will flush out the command buffer and motion planner so, at the end of jogging, the machine doesn't keep moving. Before putting time into this I'm keen to check my approach - or perhaps see that it's already supported/has a pull request out there!

My plan is to add a new code, like M409, which will do the following:
* Integrates with the emergency parser so it can skip the command queue.
* Flushes out any other commands, and flushes out the serial port.
* Flushes out all of the motions in the planner (except the currently-running one).
* Adds a stop into the planner so we can decelerate nicely.
* Waits for the motion, like M400, before continuing.

Looks like a lot of these pieces are pre-existing. M410 flushes out the motion planner, G0 can add a move into the planner, M400 will wait for the motion planner. I've also seen `GCodeQueue::clear()` to dump out any parsed commands, and `SERIAL_FLUSH()` to clear out the UART buffer. Feels like this is all the pieces I need for the above.

I'm also seen there's some new simulation features, which will probably make testing a bit easier. Are there any references for this yet? I'll admit I haven't looked much beyond coming up with my plan above.

If this seems useless / plain wrong then please let me know! I wouldn't expect this to be all that necessary for 3D printing, but will likely make CNC a bit nicer :)

### Additional context

I've seen this feature requested here already in #6532. I think my approach might differ to that used in GRBL (I haven't actually looked at their source, so I might be wrong) as I'm not trying to implement some special mode. I've assumed that I can throw in a bunch of G0 moves and then just emergency-parse and cancel them when I'm done.

dancollins, 2021-08-17:
Not much of an update, but I have figured out how to build for the simulator. It's not working for me due to OpenGL issues - as my Linux box is a headless VM - but I'll try running on my laptop and see if that works better.

Simulation is probably going to be a much more convenient way to get this feature going before I test on my MPCNC machine.

fedetony, 2021-09-20:
I suggest you turn on Realtime features, will allow you to stop and pause immediately as a CNC machine needs. Check #19253 posts.