MarlinFirmware/Marlin #22588: [BUG] Can't compile for Ender 3 V2.

lefterisgar, 2021-08-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

PlatformIO refuses to compile with the following error(s):
Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\e3v2\creality\dwin.cpp.o
Marlin\src\lcd\e3v2\creality\dwin.cpp: In function 'void Item_Adv_ProbeOffsets(uint8_t)':
Marlin\src\lcd\e3v2\creality\dwin.cpp:2429:33: error: too many arguments to function 'void say_probe_offs_en(uint8_t)'
 2429 |         say_probe_offs_en(0, row);
      |                                 ^
Marlin\src\lcd\e3v2\creality\dwin.cpp:567:6: note: declared here
  567 | void say_probe_offs_en(const uint8_t row) {
      |      ^~~~~~~~~~~~~~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\e3v2\creality\dwin.cpp.o] Error 1

### Bug Timeline

It started with commit 09af42e (17-08-2021).

### Expected behavior

I expected to be able to successfully compile Marlin firmware.

### Actual behavior

Compiling .pio\build\STM32F103RET6_creality\src\src\lcd\e3v2\creality\dwin.cpp.o fails.

### Steps to Reproduce

1. Clone Marlin.
2. Import Ender 3 V2 configurations.
3. Compile Marlin.

### Version of Marlin Firmware

Latest bugfix 2.0.x

### Printer model

Creality Ender 3 V2

### Electronics

Everything is stock.

### Add-ons

BLTouch, Filament run-out sensor

### Bed Leveling

ABL Bilinear mesh

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7001251/Marlin.zip)
[image]

I went to line 2429 of dwin.cpp and changed it from 
`say_probe_offs_en(0, row);` to `say_probe_offs_en(0);` and it did successfully compile. Is that a proper fix?

ellensp, 2021-08-18:
This PR broke it https://github.com/MarlinFirmware/Marlin/pull/22586

ellensp, 2021-08-18:
Issues resulting from Clean up CrealityUI and MarlinUI #22586

lefterisgar, 2021-08-18:
Well, thank you guys soo much!! Really appreciate your help!

lefterisgar, 2021-08-18:
Still can't get it to compile tho...

lefterisgar, 2021-08-18:
Closing was accidental.

rankinstudio, 2021-08-19:
I just ran into this bug as well, any fix?

lefterisgar, 2021-08-19:
No and I guess till the E3V2 cleanup ends we will be having a lot of issues like this...

thinkyhead, 2021-08-19:
We'll see about "a lot." Meanwhile I can quickly patch this as soon as I get back from the grocery store.

lefterisgar, 2021-08-19:
Nice! Accidents always happen and it's not your fault. Having to maintain such a big codebase is rocket science for the most of us I guess... 😁 Really appreciate your job and looking forward to the end of the cleanup!

closed 2021-08-20 by thinkyhead