MarlinFirmware/Marlin #22589: [BUG] G28 probing behaving stragelely after updating to 2.0.x (failing to raise after probing?)

LittleNyanCat, 2021-08-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying to upgrade my printer for the second time now from Marlin 1.1.9 to 2.0.x but homing is behaving very strangely and has led the nozzle crashing into [my brand new ): ] bed several times now.

Probe offsets are correct as I'm using the exact same ones I was using on 1.1.9

Most of the times after homing the printer just stays right on the bed, but still thinks the current Z height is 11.1mm, with "Endstops Z" shown on the LCD.
It sometimes only probes once and at other times probes twice (fast probe and slow probe)
It rarely raises the nozzle to the height the nozzle like it should

Manually raising Z before homing seems to have some kind of effect but I'm not sure

I've uploaded this video to show the weird behaviour:
https://youtu.be/zx55mNNQlFk

Looks like it fails to raise Z because the probe is currently triggered at that time? I have no idea...

### Bug Timeline

I tried to upgrade to 2.0.x for the first time in May last year, figured it was a bug of the branch at the time so I went back to 1.1.9. Now in 2021 it still happens.

### Expected behavior

G28 homes X & Y, double probes Z (fast probe and slow probe), then raises to the expected height, print goes on as normal

### Actual behavior

G28 homes X & Y, randomly single or double probes Z, then fails to raise while Z the firmware thinks it is raised, causing cathastrophic damage when then printer tries to go to Z = 0.15mm. Occcasionally raises to the correct height like expected

### Steps to Reproduce

1. Home the printer (either via LCD or GCode)
2. Printer ends with the nozzle on the bed despite thinking it is at Z=11.1mm

### Version of Marlin Firmware

The current bugfix branch, version 2.0.9 and 2.0.5.3

### Printer model

Anet A8 (Modded)

### Electronics

Ramps 1.4 + Reprap Discount Character LCD

### Add-ons

Servo actionated Z probe with opto trigger

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

Attached is my 1.1.9 config (that works as expected) and the 2.0.x that has issues:
[Config Marlin 1.1.9.zip](https://github.com/MarlinFirmware/Marlin/files/7000852/Config.Marlin.1.1.9.zip)
[Config Marlin 2.0.zip](https://github.com/MarlinFirmware/Marlin/files/7000851/Config.Marlin.2.0.zip)

Tannoo, 2021-08-19:
All kinds of weird things can happen when upgrading.

Did you do a factory reset? (M502 then M500)

thinkyhead, 2021-08-20:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

LittleNyanCat, 2021-08-20:
> All kinds of weird things can happen when upgrading.
> 
> Did you do a factory reset? (M502 then M500)

Yeah, the LCD asked me to after the upgrade and everything else works fine.

> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

I had tested it the day I posted this issue. Have any fixes that could be related to this been pushed in the meantime?

closed 2021-12-01 by github-actions[bot]