MarlinFirmware/Marlin #22590: [BUG] (skr1.3 dual x, sensorless homing only 1 axis is homing)

lunknowl, 2021-08-17:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

hi, i try the marlin laster firmware for skr1.3 with dual x axis using tmc2209 sensorless homing, when i jog, both axis moving fine but when i home, only 1 x axis moving? i check using dual x axis, using dual endstop, 

### Bug Timeline

maybe new

### Expected behavior

i expected when homing both x axis moving to sensorless home

### Actual behavior

only 1 x axis moving

### Steps to Reproduce

1. connect to machine
2. jog the dual x axis back and forward
3. dual x axis moving fine
4. home dual x axis, only 1 axis homing, other not moving at all

### Version of Marlin Firmware

2.0

### Printer model

Creality Ender 3

### Electronics

Stock electronics

ellensp, 2021-08-18:
Really need to see your config to see how you have set this up.
eg what pin are you using for X2 endstop?

lunknowl, 2021-08-18:
> Really need to see your config to see how you have set this up.
> eg what pin are you using for X2 endstop?

i select skr 1.3 from board list, isnt it should have all the pin for setting? the only think i do is set driver to tmc2209 and uncomment dual x with dual endstop

ellensp, 2021-08-21:
Going to presume this solved their problem, closing.

closed 2021-08-21 by ellensp