MarlinFirmware/Marlin #22592: [BUG] Cannot compile for BTT SKR 1.3 with TMC2208 drivers

MStarha, 2021-08-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

The build fails with multiple errors both with latest stable version (2.0.9.1), and bugfix. Although it reports different errors for each version.

The board is BTT SKR 1.3 with TMC2208 drivers. It compiles fine with the default A4988s.

I have attached a zip file with my configuration files as well as logs in txt files from Platform.io.

The same logs are here:

Here is a log from stable 2.0.9.1:
`
> Executing task: platformio run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1768; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1768
HARDWARE: LPC1768 100MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: 
 - framework-arduino-lpc176x 0.2.8 
 - toolchain-gccarmnoneeabi 1.90301.200702 (9.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <TMCStepper> 0.7.2
Building in release mode
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/HAL.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/HAL_MinSerial.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/HAL_SPI.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/MarlinSerial.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/eeprom_flash.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/eeprom_sdcard.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/eeprom_wired.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/fast_pwm.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/include/digipot_mcp4451_I2C_routines.c.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/include/i2c_util.c.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/main.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/tft/tft_spi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/tft/xpt2046.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/timers.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/u8g/LCD_I2C_routines.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/u8g/u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/usb_serial.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/LPC1768/watchdog.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/Delay.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/HAL_MinSerial.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/backtrace/backtrace.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/backtrace/unwmemaccess.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/eeprom_api.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/esp_wifi.cpp.o
Compiling .pio/build/LPC1768/src/src/HAL/shared/servo.cpp.o
Compiling .pio/build/LPC1768/src/src/MarlinCore.cpp.o
Compiling .pio/build/LPC1768/src/src/core/serial.cpp.o
Compiling .pio/build/LPC1768/src/src/core/utility.cpp.o
Compiling .pio/build/LPC1768/src/src/feature/ammeter.cpp.o
[... 175 lines cut ...]
Assembler messages:
Fatal error: can't create .pio/build/LPC1768/src/src/HAL/LPC1768/HAL_MinSerial.cpp.o: No such file or directory
*** [.pio/build/LPC1768/src/src/HAL/LPC1768/HAL_MinSerial.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/LPC1768/src/src/HAL/LPC1768/main.cpp.o: No such file or directory
*** [.pio/build/LPC1768/src/src/HAL/LPC1768/main.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pio/build/LPC1768/src/src/HAL/LPC1768/tft/tft_spi.cpp.o: No such file or directory
*** [.pio/build/LPC1768/src/src/HAL/LPC1768/tft/tft_spi.cpp.o] Error 1
==================================================================================== [FAILED] Took 10.75 seconds ====================================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:10.750
=============================================================================== 1 failed, 0 succeeded in 00:00:10.750 ===============================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7007627/configuration.zip)

### Version of Marlin Firmware

2.0.9.1

ellensp, 2021-08-18:
Is a issue with latest TMCStepper library 0.7.2 
https://github.com/teemuatlut/TMCStepper/issues/209

In the mean time force marlin to use 0.7.1
edit ini/features.ini  set  `HAS_TRINAMIC_CONFIG = TMCStepper@0.7.1`

MStarha, 2021-08-18:
Both stable and bugfix are already using 0.7.1

ellensp, 2021-08-18:
no they are using 0.7.1 or greater, and 0.7.2 Was just released
Note the ~ character in TMCStepper@~0.7.1

ellensp, 2021-08-18:
This is the real error error: 'uint16_t' does not name a type 14 | SSwitch(const uint16_t pin1, const uint16_t pin2, const uint8_t address);
The "Fatal error: can't create .pio/build/LPC1768/src/src/HAL/LPC1768/HAL_SPI.cpp.o: No such file or directory
*** [.pio/build/LPC1768/src/src/HAL/LPC1768/HAL_SPI.cpp.o] Error 1"
is just platformIO being daft, clicking compile again would get past it

MStarha, 2021-08-18:
Sorry about that character, I guess I was not paying enough attention.

Switching to 0.7.1 made it compile correctly for both stable and bugfix. Although bugfix needed to repeat compilation because of the same previous errors.

So shall I close this issue?

ellensp, 2021-08-18:
leave it open till library is fixed properly, so others can see this

CRCinAU, 2021-08-18:
Not sure if it makes life any easier for people, but here's a patch file that will lock to TMCStepper 0.7.1 until it can be unlocked again to follow the latest release:

```
diff --git a/ini/features.ini b/ini/features.ini
index e5b0fb3f66..b94c3fb9bc 100644
--- a/ini/features.ini
+++ b/ini/features.ini
@@ -17,7 +17,7 @@ HAS_TFT_LVGL_UI                        = lvgl=https://github.com/makerbase-mks/L
 POSTMORTEM_DEBUGGING                   = src_filter=+<src/HAL/shared/cpu_exception> +<src/HAL/shared/backtrace>
                                          build_flags=-funwind-tables
 MKS_WIFI_MODULE                        = QRCode=https://github.com/makerbase-mks/QRCode/archive/master.zip
-HAS_TRINAMIC_CONFIG                    = TMCStepper@~0.7.1
+HAS_TRINAMIC_CONFIG                    = TMCStepper@0.7.1
                                          src_filter=+<src/feature/tmc_util.cpp> +<src/module/stepper/trinamic.cpp> +<src/gcode/feature/trinamic/M122.cpp> +<src/gcode/feature/trinamic/M906.cpp> +<src/gcode/feature/trinamic/M911-M914.cpp>
 HAS_STEALTHCHOP                        = src_filter=+<src/gcode/feature/trinamic/M569.cpp>
```

teemuatlut, 2021-08-18:
I'll push a fix in a bit.
If you want to change Marlin side then I'd always prefer [blacklisting a certain release](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html) with `TMCStepper@~0.7.1,!=0.7.1` as that won't prevent you from updating in the future.

CRCinAU, 2021-08-18:
> I'll push a fix in a bit.

Awesome, thanks.

> If you want to change Marlin side then I'd always prefer [blacklisting a certain release](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html) with `TMCStepper@~0.7.1,!=0.7.2` as that won't prevent you from updating in the future.

Oh yeah - I certainly wouldn't even think about putting this into the main repo - however its a workaround to allow people to build right now.

thisiskeithb, 2021-08-18:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/c04b2af3bf38490be423a356787890ada677b2ee and:

> v0.7.3 is pushed now and I'd expect PIO to pick it up in a day or two.

closed 2021-08-18 by thisiskeithb