MarlinFirmware/Marlin #22593: [BUG] M48 is broken in the latest update

eblieb, 2021-08-18:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

M48 will probe twice and then send a Start command and not continue. Reverted to a firmware compiled 20 days ago and it works on that firmware. 

### Version of Marlin Firmware

02000901

### Printer model

Ender 3 V2

eblieb, 2021-08-18:
https://github.com/Jyers/Marlin/tree/v1.3.5.b - Was running this fork of Marlin

Vertabreak, 2021-08-18:
fork are generally not supported here as we have no control over code on forks.

thisiskeithb, 2021-08-18:
We can't support forks of Marlin here.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-18 by thisiskeithb