MarlinFirmware/Marlin #22596: [QUESTION] does ENABLE_LEVELING_AFTER_G28 disable M420 S0?

digant73, 2021-08-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

It seems that if ENABLE_LEVELING_AFTER_G28 is enabled then ABL cannot be disabled with M420 S0.
Is that the intended behavior or is it a bug?
The BTT TFT fw is based on temporary disabling ABL on menus such as Probe Offset, Mesh Edit etc... So with ENABLE_LEVELING_AFTER_G28  enabled M420 S0 has no effect and wrongs results are obtained.
If it's a bug, please fix it

### Expected behavior

M420 S0 disables ABL even with ENABLE_LEVELING_AFTER_G28 enabled

### Actual behavior

M420 S0 doesn't disable ABL if ENABLE_LEVELING_AFTER_G28 is enabled

### Version of Marlin Firmware

2.0.9.1

### Bed Leveling

UBL Bilinear mesh

borland1, 2021-08-19:
You can try configuring `Configuration_adv.h` by defining startup commands `M420 S0`    That should enable auto/manual bed leveling on startup, and the setting can be manually overrided using the LCD menu.

digant73, 2021-08-19:
Ok thanks, but it seams to me more to try a workaround attempt than an answer to my question.
According to description reported in Configuration.h ENABLE_LEVELING_AFTER_G28 should only enable ABL after G28 while 
RESTORE_LEVELING_AFTER_G28 restore the previous ABL state.  A use RESTORE_LEVELING_AFTER_G28 without any problem (it is possible to disable and re-enable ABL after G28

thinkyhead, 2021-08-19:
I can think of no reason why `M420 S0` command sent after a `G28` would not disable bed leveling.

If some serial controllers are expecting certain results by sending G-code without also including the appropriate `M420 Sn` in coordination with other commands, then the configurations with such controllers enabled should probably not be enabling `ENABLE_LEVELING_AFTER_G28`, `RESTORE_LEVELING_AFTER_G28`, or other commands that interfere with their requirements.

digant73, 2021-08-20:
I think so. BTT TFT fw is based on always disabling ABL (M420 S0) to perform tasks such as Z probe offset, mesh editing. Also gcode files often provide M420 Sx just to instruct Marlin to use the desired BL (enabled or disabled).
With RESTORE_LEVELING_AFTER_G28 enabled, M420 S0 is invalidated (it doesn't disable ABL) so the serial controller cannot do what it wants to do. As an example, BTT TFT fw disables ABL (and endstops) for mesh editing. The nozzle is moved on the mesh XY coord and then the nozzle is moved to the Z coord including the related z-height (e.g. if z-height is -0.4 then the nozzle should be moved to -0.4). With ABL not disabled the nozzle will be moved to Z-0.8 (twice the z-height) while Marlin will still continue to provide coord Z-0.4 to the TFT (the TFT is not aware at all that ABL remained engaged)

digant73, 2021-08-22:
forget it. I found the problem on the TFT fw and fixing it just now.

closed 2021-08-22 by digant73