MarlinFirmware/Marlin #22597: [BUG] LCD Menu Halts Printer

AfroPsycho, 2021-08-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Hey,
I am trying to use Marlin 2.0.9.1
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/7016331/Configuration.zip)
 on my MKS Gen V1.4 Motherboard 
I did compile the sketch and upload it it works but with a problem
When ever I try to press encoder button printer halts 
I know it is not a problem with encoder because it works with marlin 1.1.x and also when I first upload marlin 2.0 there was a problem with the EEPROM and I can choose and click to reset option using encoder 
So what I assume is something in the lcd menu crashes the printer and I can't open the lcd menu

BTW using pronterface I can do basic stuff like G28, G29, G1 X100 
I didn't try anything further yet 
What could be the possible reason of the problem and how can I solve it 
I have uploaded the Configuration.h & Configuration_adv.h

### Expected behavior

To open LCD Menu

### Actual behavior

Printer Halted 

### Version of Marlin Firmware

2.0.9.1

### Printer model

3D4E 4GenPro 2R

### Electronics

MKS Gen 1.4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

ellensp, 2021-08-20:
Try adding `#define KILL_PIN             -1`  to your Configuration.h

borland1, 2021-08-22:
This bug report may be related to reported issue number 21109, although symptoms appear different.

AfroPsycho, 2021-08-25:
> Try adding `#define KILL_PIN -1` to your Configuration.h

this kind of solved the issue but now I have a new problem
pressing the encoder button no longer halts the printer but now pressing it does nothing 
like I said before I did use the encoder when printer halted because of EEPROM but just thinking about it it may killed the printer whenever I pressed it

I assume right now my problem is with my encoders wiring

ellensp, 2021-08-25:
yes it would seem that your BTN_ENC is wired to EXP2_03_PIN when it should be on EXP1_09_PIN 

```
/**
 * LCD adapter. Please note: These comes in two variants. The socket keys can be
 * on either side, and may be backwards on some boards / displays.
 *         -----                           -----
 *    D37 |10 9 | D35          (MISO) D50 |10 9 | D52 (SCK)
 *    D17 | 8 7 | D16                 D31 | 8 7 | D53
 *    D23   6 5   D25                 D33   6 5   D51 (MOSI)
 *    D27 | 4 3 | D29                 D49 | 4 3 | D41
 *    GND | 2 1 | 5V                  GND | 2 1 | NC
 *         -----                           -----
 *         EXP1                            EXP2
 */
```

AfroPsycho, 2021-08-25:
#define BTN_EN1

> yes it would seem that your BTN_ENC is wired to EXP2_03_PIN when it should be on EXP1_09_PIN
> 
> ```
> /**
>  * LCD adapter. Please note: These comes in two variants. The socket keys can be
>  * on either side, and may be backwards on some boards / displays.
>  *         -----                           -----
>  *    D37 |10 9 | D35          (MISO) D50 |10 9 | D52 (SCK)
>  *    D17 | 8 7 | D16                 D31 | 8 7 | D53
>  *    D23   6 5   D25                 D33   6 5   D51 (MOSI)
>  *    D27 | 4 3 | D29                 D49 | 4 3 | D41
>  *    GND | 2 1 | 5V                  GND | 2 1 | NC
>  *         -----                           -----
>  *         EXP1                            EXP2
>  */
> 
>       #define BTN_EN1              EXP2_08_PIN
>       #define BTN_EN2              EXP2_06_PIN
>       #define BTN_ENC              EXP1_09_PIN
>       #define KILL_PIN             EXP2_03_PIN
> ```

Well I can't solve this 
it is so complicated compared to Marlin 1.1.x
I have the manufacturers pins file for Marlin 1.1.x so basically I know which pins are used for encoder but I can't compare it to marlin 2.0.x
Would you help me with that I am uploading the file 

I am using REPRAP_DISCOUNT_SMART_CONTROLLER
[pins_MKS_GEN_V_1_4.zip](https://github.com/MarlinFirmware/Marlin/files/7046720/pins_MKS_GEN_V_1_4.zip)

ellensp, 2021-08-25:
The provided pins files also says that BTN_ENC is on pin D35 (EXP1_09_PIN) just like current marlin.

Is the lcd connected with individual wires you can change? or it it just two 10 pin connectors?

ellensp, 2021-08-25:
If its the latter add this custom fix to end of the file Marlin/src/pins/ramps/pins_MKS_GEN_13.h

```
#undef BTN_ENC
#undef KILL_PIN
#define BTN_ENC                     EXP2_03_PIN
#define KILL_PIN                    -1

```

AfroPsycho, 2021-08-25:
well it is 2x10 pin connector and because it is so hard to rewire it I did add that custom fix and it did solve the issue 
Thanks for help

AfroPsycho, 2021-08-25:
but still there is a question mark 
if it is wired correctly why this problem occurs

ellensp, 2021-08-25:
your LCD screen seems to be non standard

ellensp, 2021-08-25:
There is nothing Marlin can do about that

closed 2021-08-25 by ellensp