MarlinFirmware/Marlin #22598: [BUG] M115 - badly reported print area dimensions.

radek8, 2021-08-19:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a printing area of 230x230
The end stops are in positions X -12 and Y -33
The Marlin defines:

// The size of the printable area
#define X_BED_SIZE 230
#define Y_BED_SIZE 230

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS -12
#define Y_MIN_POS -33
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 255

The M115 command reports incorrect pad dimensions

The work surface (print bed) does not start at position -12 and -33 but at position 0.

### Expected behavior

Therefore, the response to the M115 command should look like this correctly:
area: {full: {min: {x: -12.0000, y: -33.0000, z: 0.0000}, max: {x: 230.0000, y: 230.0000, z: 255.0000}}, work: {min: {x: 0.0000 , y: 0.0000, z: 0.0000}, max: {x: 230.0000, y: 230.0000, z: 255.0000}}}

the workspace does not start at position -12 and -33 but at position 0.

### Actual behavior

The answer to the M115 command is:

area: {full: {min: {x: -12.0000, y: -33.0000, z: 0.0000}, max: {x: 230.0000, y: 230.0000, z: 255.0000}}, work: {min: {x: - 12.0000, y: -33.0000, z: 0.0000}, max: {x: 230.0000, y: 230.0000, z: 255.0000}}}

### Version of Marlin Firmware

Marlin bugfix 2.0.x - from 14.8.2021

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/7017539/Config.zip)

thinkyhead, 2021-08-19:
Try this in `M115` and see if it produces a better result. Check it with soft endstops enabled/disabled and with `G92` workspace offsets (as applied to `CNC_COORDINATE_SYSTEMS`). If it looks correct then I can merge this change soon.
```cpp
    #if ENABLED(M115_GEOMETRY_REPORT)
      const xyz_pos_t bmin = { 0, 0, 0 },
                      bmax = { X_BED_SIZE , Y_BED_SIZE, Z_MAX_POS },
                      dmin = { X_MIN_POS, Y_MIN_POS, Z_MIN_POS },
                      dmax = { X_MAX_POS, Y_MAX_POS, Z_MAX_POS };
      xyz_pos_t cmin = bmin, cmax = bmax;
      apply_motion_limits(cmin);
      apply_motion_limits(cmax);
      const xyz_pos_t lmin = dmin.asLogical(), lmax = dmax.asLogical(),
                      wmin = cmin.asLogical(), wmax = cmax.asLogical();
      SERIAL_ECHOLNPAIR(
        "area:{"
          "full:{"
            "min:{x:", lmin.x, ",y:", lmin.y, ",z:", lmin.z, "},"
            "max:{x:", lmax.x, ",y:", lmax.y, ",z:", lmax.z, "}"
          "},"
          "work:{"
            "min:{x:", wmin.x, ",y:", wmin.y, ",z:", wmin.z, "},"
            "max:{x:", wmax.x, ",y:", wmax.y, ",z:", wmax.z, "}",
          "}"
        "}"
      );
    #endif
```

radek8, 2021-08-19:
thank you, now it works as expected

area:{full:{min:{x:-12.0000,y:-33.0000,z:0.0000},max:{x:230.0000,y:230.0000,z:255.0000}},work:{min:{x:0.0000,y:0.0000,z:0.0000},max:{x:230.0000,y:230.0000,z:255.0000}}}

closed 2021-08-19 by thinkyhead

radek8, 2021-08-20:
The bed geometry report is OK, but isn't there a problem moving the head to the extreme position?
I think the line
xyz_pos_t cmin = bmin, cmax = bmax;
should remain in its original shape
xyz_pos_t cmin = dmin, cmax = dmax;

I have 2 nozzles the second nozzle has an offset of -12mm.
Therefore, the end stop had an offset of -12mm so that the edge of the bed was also available for the second nozzle.
Will the left edge now be available for the second nozzle after G28?
I'll test it tonight.

radek8, 2021-08-20:
Tested even when printing the second nozzle in the extreme position.
It works well the way you fixed it.
Thank you

commit 033e65ec (2021-08-19): 🐛 Show bed size as 'work:' in M115

Fixes #22598
commit caa6ec05 (2021-10-03): 🐛 Show bed size as 'work:' in M115

Fixes #22598