MarlinFirmware/Marlin #22601: [BUG] Ender 3 V2 DWIN display missing icons

lefterisgar, 2021-08-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I updated the DWIN_SET on my Ender 3 V2 and flashed the newest bugfix of Marlin, the display has a few missing icons. On the main screen where it says "Print", "Control", "Prepare" and "Info" all icons show up okay. But if I try to go to a submenu there is not a single icon in here.

### Bug Timeline

It started with the CrealityUI cleanup.

### Expected behavior

I expected all icons to load when I navigate submenus, e.t.c. Also, when the printer boots up, I expected to be able to see the loading progress bar.

### Actual behavior

- No progress bar when the printer boots up.
- No icons on any submenu except the main screen with 4 icons.
- No icon on bed leveling or anything else.
- Text sometimes appears corrupted.
- The display is a bit more sluggish than it used to be.

### Steps to Reproduce

1. Update the Ender 3 V2 display with the latest DWIN_SET (bugfix-2.0.x of Marlin/Configurations).
2. Compile the latest bugfix-2.0.x branch of Marlin and flash it to the printer.
3. Have a look of what I mentioned above.
4. Compile / flash the latest version of Jyer's Marlin or Creality's stock version.
5. Verify that the screen works properly.

### Version of Marlin Firmware

Latest bugfix 2.0.x

### Printer model

Creality Ender 3 V2

### Electronics

Everything is stock.

### Add-ons

BLTouch, Filament run-out sensor, Marlin Binary File Transfer Protocol

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/7021758/Configs.zip)

ellensp, 2021-08-22:
I just tried the current E3V2 bugfix configs and DWIN_SET, I don't get missing icons.
The  only change I made was the motherboard (BOARD_CREALITY_V427 vs a V4)

ellensp, 2021-08-22:
Can you try flashing the LCD and Controller again with latest bugfix?

lefterisgar, 2021-08-22:
Yes I agree that it's a weird problem. FYI after re-flashing everything (already had the latest DWIN) it is now fixed. This problem occured to me at the same time as #22588. But anyways now it's up again! Thanks a lot @ellensp!

closed 2021-08-22 by lefterisgar

phishsticks, 2021-09-15:
I have this issue too, have reflashed multiple times with always the same result.

lefterisgar, 2021-09-18:
Are you sure the SD card is formatted the right way?