MarlinFirmware/Marlin #22606: 2.0.9.1 Will Not Compile

rq3, 2021-08-20:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am currently running Marlin 2.0.9

When I download and unzip 2.0.9.1, and install the configuration.h and configuration_adv.h files from the config/examples/delta/anycubic/predator folder, it will not compile under Platformio under VSCode.

It throws this problem:
Cannot find "C:\Users\rquin\.platformio\packages\framework-arduinoststm32@4.10900.200819\variants\MARLIN_F103Zx".

Marlin 2.0.9 compiles fine.

### Bug Timeline

Just noticed today.

### Expected behavior

I expect Marlin 2.0.9.1 to compile.

### Actual behavior

Marlin 2.0.9.1 fails to compile, and is looking for a non-existent file.

### Steps to Reproduce

1). Download Marlin 2.0.9.1 and unzip it.
2) Download the configuration files from the /config/examples/delta/Anycubic/Predator/ folder
3) Install those files in the Marlin directory
4) Modify platformio.ini to the trigorilla_pro
5) Attempt to compile

### Version of Marlin Firmware

2.0.9.1

### Printer model

Anycubic Predator

### Electronics

Stock

### Add-ons

none

ellensp, 2021-08-21:
Build env: trigorilla_pro_maple works, but not Build env:trigorilla_pro

Give this a try
Edit ini/stm32f1.ini
find 
```CPP
[env:trigorilla_pro]
platform      = ${common_stm32.platform}
extends       = env:mks_robin
extra_scripts = ${common_stm32.extra_scripts}
```
replace with
```CPP
[env:trigorilla_pro]
platform      = ${common_stm32.platform}
extends       = env:mks_robin
extra_scripts = ${stm32_variant.extra_scripts}
```
Also edit Marlin/src/pins/stm32f1/pins_TRIGORILLA_PRO.h
Find
```CPP
#define FSMC_RS_PIN                       PD11  // A0
```
Change it to
```CPP
  #define FSMC_RS_PIN                       PD11  // A0

  #define TFT_CS_PIN                 FSMC_CS_PIN
  #define TFT_RS_PIN                 FSMC_RS_PIN
```

This compiles, but I have no way to test it

rq3, 2021-08-21:
With your changes, and making serial port 1=1 instead of 0, and enabling serial port 2 and assigning it 3, it compiles. I will try it on the machine.

rq3, 2021-08-21:
I just tried it on the machine. No response at all with completely dead display. Interesting to note that the compile yields two bin files: firmware.bin (77KB), and robin.bin (77KB).

A "normal" firmware.bin for this machine is about 243KB.

thinkyhead, 2021-08-21:
You can also try this, which avoids borrowing from Robin but uses similar flags…
```ini
[env:trigorilla_pro]
platform             = ${stm32_variant.platform}
extends              = stm32_variant
board                = genericSTM32F103ZE
board_build.variant  = MARLIN_F103Zx
board_build.offset   = 0x7000
build_flags          = ${stm32_variant.build_flags}
                       -DENABLE_HWSERIAL3 -DTIMER_SERIAL=TIM5
build_unflags        = ${stm32_variant.build_unflags}
                       -DUSBCON -DUSBD_USE_CDC
```

I note that the Maple build uses these build flags: `-DSS_TIMER=4 -DSTM32_XL_DENSITY` so possibly the build flags above that I borrowed from `env:mks_robin` are not entirely correct. We could use advice from someone having more familiarity with the Trigorilla Pro board.

thinkyhead, 2021-08-21:
I have posted the PR linked above which allows it to compile. Let's continue the conversation and figure out if any more changes will be needed over on that page.

closed 2021-08-21 by thinkyhead

rq3, 2021-08-21:
@thinkyhead ; With your latest ini changes, it compiles fine with no errors or warnings, and the result is just one firmware.bin of 77KB (normally 243KB).

But same result in the machine. The display is totally dead.

thinkyhead, 2021-08-22:
Strange. When I did the build it came out as the normal size. Maybe try again…?

PR #22610 (2021-08-30): Fix Trigorilla Pro for HAL/STM32 build
Fixes #22606

Patches another glitch from the updated STM32 build environments. The build flags are borrowed from the MKS Robin environment, as inherited from the Maple environments. Due to differences in the platforms it is unclear whether the build flags are entirely correct.