MarlinFirmware/Marlin #22613: [BUG] (bug summary)

Dragon41673, 2021-08-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to add a BIQU Filament Break Sensor (non-smart model) while filament is IN the sensor, the sensor tells the printer that filament is NOT in the sensor. When the filament is removed, it tells the printer that filament is IN the sensor. Is it possible to add a check box to reverse the read on the sensor to resolve this issue?

### Bug Timeline

New to me, just discovered a couple days ago

### Expected behavior

Provide a check box to change the sensor reading to allow proper reading of the filament sensor

### Actual behavior

Filament sensor reads wrong. When filament is inserted, printer reads it as the filament has run out. When filament is NOT in the sensor, printer reads it as there is filament in the sensor.

### Steps to Reproduce

1. Filament sensor is the basic BIQU Filament Break Sensor found here...
https://www.amazon.com/dp/B08QJDP36R/ref=dp_iou_view_product?ie=UTF8&psc=1

2. Just insert filament and you will see

### Version of Marlin Firmware

V1.3.4

### Printer model

Ender 3 V2

### Electronics

Stock

### Add-ons

Creality Direct Drive Extruder & BL Touch Sensor

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

None

thisiskeithb, 2021-08-22:
> Version of Marlin Firmware
> V1.3.4

Sounds like you’re running Jyer’s fork of Marlin, which we can’t support here.

> Provide a check box to change the sensor reading to allow proper reading of the filament sensor

You will need to adjust `FIL_RUNOUT_STATE`.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-08-22 by thisiskeithb

Dragon41673, 2021-08-22:
Ugh...yeah, I am. I thought this was his Github, sorry...first time trying to do this here and it's not exactly the easiest to understand for new people