MarlinFirmware/Marlin #22615: [BUG] (Compilation error when enable CHAMBER_FAN)

diogoc, 2021-08-22:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When enable the CHAMBER_FAN without enable the AUTO_FAN it gives a compilation error CHAMBER_FAN_INDEX not defined.

The chamber fan is limited to 225.

### Expected behavior

The chamber fan should not need the auto fan to work.

Max value of chamber fan should be 255.

### Actual behavior

Gives a compilation error CHAMBER_FAN_INDEX not defined.

NOMORE(fan_chamber_pwm, 225) in temperature.cpp should be NOMORE(fan_chamber_pwm, 255)

### Version of Marlin Firmware

Marlin Bugfix-2.0.x 2021-08-21

### Additional information & file uploads

It would be good if it was possible to configure the pin used for the chamber fan.

ladismrkolj, 2021-12-05:
I looked into the CHAMBER_FAN_INDEX
I added some code into the temperature.cpp to fix it. If it works as expected I will do a pull request.

thisiskeithb, 2022-04-01:
https://github.com/MarlinFirmware/Marlin/pull/23262 has been merged.

closed 2022-04-01 by thisiskeithb