MarlinFirmware/Marlin #22616: [BUG] FAN_PIN does not bind to

Quas7, 2021-08-22:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

Firstly, I did not yet test with bugfix-2.0.x but will do in the coming days.
I use here an old DUE+RADDS combo and wanted to add a part cooling fan recently as on my other printers.

```
#undef FAN_PIN   //RADDS has 9 as standard
#define FAN_PIN         8
#undef FAN1_PIN //RADDS has 8 as standard
#define FAN1_PIN         -1
#define E0_AUTO_FAN_PIN 9
```

M43 output:
```
Recv: PIN: 08        FAN_PIN                                protected
Recv: .              FAN1_PIN                               protected
Recv: PIN: 09        E0_AUTO_FAN_PIN                        protected
```
looking all as expected.

### Expected behavior

I expect that "M106 S255" is starting FAN_PIN with max PWM.

### Actual behavior

But "M106 S255" does not start the part cooling fan (does nothing).
On the other hand M106 **P1** S255 does start it but FAN1_PIN is unbound.

### Steps to Reproduce

see config entries above

### Version of Marlin Firmware

02000901

### Printer model

FT5 with RADDS+DUE

### Electronics

RADDS+DUE

### Add-ons

a few unrelated

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/7028273/config.zip)

ellensp, 2021-08-22:
"FAN1_PIN is unbound" is not unbound, it is attached pin 8 also.

Your not interpreting M43 correctly.
```
Recv: PIN: 08        FAN_PIN                                protected
Recv: .              FAN1_PIN                               protected

```
means that FAN1_PIN is also on pin 8

ellensp, 2021-08-22:
Adding the following to Configuration_adv.h is not sufficient
```
#undef FAN_PIN
#define FAN_PIN         8
#undef FAN1_PIN
#define FAN1_PIN         -1

```

The pins file Marlin/src/pins/sam/pins_RADDS.h always supersedes this.
Unless it has been coded not to
It has the following.
```
#ifndef FAN_PIN
  #define FAN_PIN                              9
#endif
#define FAN1_PIN                               8
```

Ie adding #define FAN_PIN to Configuration.h or advance will work fine.
but FAN1_PIN will be over written.

You need to edit your pins_RADDS.h
change 
`#define FAN1_PIN                               8
`to
```
#ifndef FAN1_PIN
  #define FAN1_PIN                              8
#endif
```

And you dont need either of the 
```
#undef FAN_PIN
#undef FAN1_PIN

```

closed 2021-08-22 by ellensp