MarlinFirmware/Marlin #22622: [BUG] Extruder stops working during printing

Diogo24P, 2021-08-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Good Morning
I'm having a problem with my 2.0.9.1 firmware.
When a print starts, the extruder engine stops working, it may be at the end of the print, as in the beginning, there is no right time.
It only goes back to work if it pauses printing and then resumes, it then works until the printer is turned off.
When I start the printer, I have to start printing, pause and resume to print without stopping the extruder motor.
I tried with another version 2.0.9.1 and didn't have this problem with the same g-code file.

What is generating this? 

[Marlin-2.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/7038588/Marlin-2.0.x.zip)

### Expected behavior

Work without stopping .

### Steps to Reproduce

Turn on the printer.
Start 1 impression .

### Version of Marlin Firmware

2.0.9.1

### Printer model

Sapphire Plus

### Electronics

Extruder orbiter or original extruder, same problem in 2, Moquito Clone, Bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Simplify3D

### Host Software

SD Card (headless)

### Additional information & file uploads

I recently installed the Extruder orbiter, but even before installing this extruder, I already have this problem.

ellensp, 2021-08-24:
LIN_ADVANCE is incompatible with TMC2208_STANDALONE. They randomly crash. These stepper driver cannot be switched into spreadcycle mode and this is required for LIN_ADVANCE.
You need none standalone versions (and a controller that can use non standalone mode)

closed 2021-08-24 by ellensp

Diogo24P, 2021-08-24:
The tms are 2225.
LIN_ADVANCE is already disabled

ellensp, 2021-08-24:
Provided configs says 
#define X_DRIVER_TYPE  TMC2208_STANDALONE
#define Y_DRIVER_TYPE  TMC2208_STANDALONE
#define Z_DRIVER_TYPE  TMC2208_STANDALONE
#define Z2_DRIVER_TYPE TMC2208_STANDALONE
#define E0_DRIVER_TYPE TMC2208_STANDALONE
and that LIN_ADVANCE is enabled