MarlinFirmware/Marlin #22623: [BUG] ABL not compensating enough.

lolgamer2005, 2021-08-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

None of the ABL systems seem to work on my new hardware, I have installed an SKR 2 from bigtreetech with a genuine BlTouch 3.1, but Z compensation doesn´t move enough, even though the bed is almost leveled.

### Bug Timeline

Its a new bug I have came across when I changed the hardware on my anet a6.

### Expected behavior

I expected it to generate a mesh and use it to compensate and print a great first layer.

### Actual behavior

It "works" but not fine. It compensates, but it is far from doing the amount of compensation it should, so it ends up crashing the bed on one side and not stisking on the other.

### Steps to Reproduce

Install an skr 2 and a bltouch 3.1 with a frankenstein carriage (original design in thingiverse) on an anet a6.
Set up any kind of ABL system on marlin.
Try to print a 1 layer square and see the result.

### Version of Marlin Firmware

Lastest version 2.0.9.1

### Printer model

Anet A6

### Electronics

Upgraded Skr 2 board from bigtreetech.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Same as my slicer

### Additional information & file uploads

It doesnt let me upload the .zip :(

DerAndere1, 2021-08-24:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

    Download Marlin bugfix-2.0.x to test with the latest code.
    Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
    Connect to your printer from host software such as Cura, Printrun or Repetier Host.
    Send M502 and M500 to ensure your Configurations are applied.
    Issue the command M111 S247 to enable maximum logging.
    Perform a G28 to do your standard homing procedure.
    Do a G29 to probe the bed. This will also enable bed leveling.
    Copy the log output into a .TXT file and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

radek8, 2021-08-24:
Hi,
I have the same problem. I installed the latest Marlin Bugfix 2.0.x (from today)
I warm the bed, level the bed with G29.
I'll start printing across the bed. print on the right side is good, nozzle about 0.2 mm above the pad.
As the nozzle moves to the left, the gap between the nozzle and the bed decreases. On the left side of the bed, the nozzle is already touching the bed.
[Debug.txt](https://github.com/MarlinFirmware/Marlin/files/7042012/Debug.txt)
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7042016/Marlin.zip)

[image]

DerAndere1, 2021-08-24:
Thanks. The original report says it happens with any ABL method. Note, that for AUTO_BED_LEVELING_BILINEAR, there are already open bug reports (e.g. https://github.com/MarlinFirmware/Marlin/issues/22301) that might interfere with this general bug. So it would be especially helpful to see the log file from a run where you used a different ABL method and observed the bug.

radek8, 2021-08-24:
I have a problem with AUTO_BED_LEVELING_BILINEAR.
I haven't tried another method.
So are you proposing to test another method, such as UBL?

thinkyhead, 2021-08-24:
Because bed leveling is an extremely straightforward and highly tested procedure, I think that all these bugs about bed leveling "not being applied" are incorrect and are the result of mechanical problems with your printers. Specifically, your probe is not staying perfectly perpendicular to the bed throughout the entire probing procedure, leading to readings being too high on one side and too low on the other. Since the mesh data is garbage there is nothing we can do to compensate for it. You must fix your mechanical issues.

You can confirm for yourself whether your probe is messed up by using Manual Probing and then comparing the resulting mesh to the one you get from Automatic Probing. If you find that the Manual Mesh is less "tilted" than your automatic probe result, this is a clear indication of a mechanical problem.

To fix your mechanical issue, make sure your probe mounting is well-tightened and make sure that there are no cables pulling on the probe as the gantry moves in the X axis. If you have a printer with a carriage that rides on wheels, you may have a twisted gantry, which ends up causing the same problem of garbage readings.