MarlinFirmware/Marlin #22626: [FR] build for Anet V1.0 / Anet V1.5 board with optiboot bootloader

DerAndere1, 2021-08-24:
### Is your feature request related to a problem? Please describe.

With my electronics, either comilation or upload fails, depending on what default_env I use. 

### Are you looking for hardware support?

Anet V1.0 board (actually, the board has "Anet V1.5" printed on it) with optiboot bootloader

### Describe the feature you want

Ability to build + upload Marlin using VS Code according info in pins_ANET_10.h.

Currently, optiboot requires one of the following workarounds:

**Alternative A:**

WARNING: This workaround is not intended for users with the stock bootloader. It may corrupt the stock bootloader

https://github.com/MarlinFirmware/Marlin/blob/7974dcd2aaae741e37d6e2e230492f2ba258ffb4/Marlin/src/pins/pins.h#L332 change to
```
  #include "sanguino/pins_ANET_10.h"            // ATmega1284P                            env:sanguino1284p env:sanguino1284p_optimized env:melzi_optiboot
```

For consistancy, https://github.com/MarlinFirmware/Marlin/blob/7974dcd2aaae741e37d6e2e230492f2ba258ffb4/Marlin/src/pins/sanguino/pins_ANET_10.h#L90 change to
  #error "Oops! Select 'Sanguino' in 'Tools > Board' and 'ATmega1284P' in 'Tools > Processor.' (For PlatformIO, use 'sanguino1284p', or 'sanguino1284p_optimized'. With optiboot, use 'melzi_optiboot')."

https://github.com/MarlinFirmware/Marlin/blob/7974dcd2aaae741e37d6e2e230492f2ba258ffb4/Marlin/src/pins/sanguino/env_validate.h#L37 change to
  #error "Oops! Select 'Sanguino' in 'Tools > Board' and 'ATmega1284P' in 'Tools > Processor.' (For PlatformIO, use 'sanguino1284p', or 'sanguino1284p_optimized'. With optiboot, use 'melzi_optiboot')."

This allows the user to build and upload after setting `default_env = melzi_optiboot`  in platformio.ini .

**Alternative B:**

WARNING: This workaround is not intended for users with the stock bootloader. The firmware upload may fail with the stock bootloader.

https://github.com/MarlinFirmware/Marlin/blob/7974dcd2aaae741e37d6e2e230492f2ba258ffb4/ini/avr.ini#L103-L107 change to:
```
[env:sanguino1284p]
platform                  = atmelavr
extends                   = common_avr8
board                     = sanguino_atmega1284p
board_upload.maximum_size = 126976            # TODO: This may be increased slightly when using optiboot
upload_speed        =  115200
```

For consistancy, https://github.com/MarlinFirmware/Marlin/blob/7974dcd2aaae741e37d6e2e230492f2ba258ffb4/Marlin/src/pins/sanguino/pins_ANET_10.h#L90 change to
  #error "Oops! Select 'Sanguino' in 'Tools > Board' and 'ATmega1284P' in 'Tools > Processor.' (For PlatformIO, use 'sanguino1284p', or 'sanguino1284p_optimized'. With optiboot, use 'sanguino1284p' and set `change upload_speed =  115200`)."

https://github.com/MarlinFirmware/Marlin/blob/7974dcd2aaae741e37d6e2e230492f2ba258ffb4/Marlin/src/pins/sanguino/env_validate.h#L37 change to
  #error "Oops! Select 'Sanguino' in 'Tools > Board' and 'ATmega1284P' in 'Tools > Processor.' (For PlatformIO, use 'sanguino1284p', or 'sanguino1284p_optimized'. With optiboot, use 'sanguino1284p' and verify that `upload_speed = 115200`)."

This allows the user to build and upload after setting `default_env = sanguino1284p` in platformio.ini . 

### Additional context

the melzi_optiboot option was removed for the Anet V1.0 board in PR https://github.com/MarlinFirmware/Marlin/pull/17120 to prevent users with stock bootloader to exceed firmware size limit of 12697 bytes.

DerAndere1, 2021-08-24:
@kreuzhofer as the author of the cited pull request, do you have a preferred build procedure for the Anet V1.0 board with optiboot bootloader ?

thinkyhead, 2021-08-24:
For an alternative build with a different start address or other modifications, you must add a new environment entry to the `avr.ini` file.

DerAndere1, 2021-08-24:
I found this https://docs.platformio.org/en/latest/boards/atmelavr/sanguino_atmega1284p.html#configuration which says

> You can override default Sanguino ATmega1284p (16MHz) settings per build environment using board_*** option, where *** is a JSON object path from board manifest sanguino_atmega1284p.json. For example, board_build.mcu, board_build.f_cpu, etc.

So to overwrite the start address, we would need `board_build.upload.offset_address = 0x...` in the `env` . Question is, how  `melzi_optiboot` worked without such setting

DerAndere1, 2021-08-25:
The reason why melzi_optiboot was disallowed for anet 1.0 in PR #17120 does not exist anymore since PR https://github.com/MarlinFirmware/Marlin/pull/18502 . But since PR https://github.com/platformio/platform-atmelavr/pull/217 was merged into PlatformIO, the board file specifies the complete flash size change (`maxiumsize = 131072,`). Since then, the `melzi_optiboot` environment was probably broken for all boards: We now need to have `board_upload.maximum_size = ...` in all `env`s to account for the bootloader size. I edited my proposal in the initial request and will prepare a PR. I think we should edit pins.h to allow users with Anet 1.0 board to set `default_env = melzi_optiboot` because the name of the environment makes its purpose obvious, especially if we document everything in the pins_ANET_10.h file.