MarlinFirmware/Marlin #22629: MKS assets doens't update with MultiVolume

DvoraNoob, 2021-08-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm trying update mks assets with Multi_Volume, but the firmware doesn't read the folder "assets", but with multi_volume off, works fine

### Steps to Reproduce

1. set marlin for mks robin nano v3
2.  #define MULTI_VOLUME and #define USB_FLASH_DRIVE_SUPPORT and #define USE_OTG_USB_HOST
3. compile
4. put bin archieve and folder "assets" to SD root
5. upload to printer
6. the firmware load but the assets doesn't 

### Version of Marlin Firmware

Marlin bugfix-2.0.x #define CONFIGURATION_H_VERSION 02000901

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

mks-viva, 2021-09-13:
@DvoraNoob 
Can you provide source code to me and i test it?

DvoraNoob, 2021-09-13:
> @DvoraNoob
> Can you provide source code to me and i test it?

Has 2 pastes, one with no edit and another with edit for MKS Robin Nano V3+TMC2209+WIFI 

https://1drv.ms/u/s!AlJymIpivUgtmnhdoOSuC0FCzr87?e=8PcdF9

closed 2021-11-23 by github-actions[bot]