MarlinFirmware/Marlin #22653: [BUG] Z Probe Offset not applied homing using Z min endstop instead of a probe (BLTouch)

digant73, 2021-08-28:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When homing using Z min endstop sensor (USE_PROBE_FOR_Z_HOMING and Z_SAFE_HOMING disabled) the Z probe offset (I set it also using the Z probe wizard) is never applied.
When homing using the probe (USE_PROBE_FOR_Z_HOMING and Z_SAFE_HOMING enabled) the Z probe offset is properly applied.

NOTES:
1) the probe is connected and configured (on my RAMPS board) on Z max endstop (#define Z_MIN_PROBE_PIN 19)
2) homing with Z min endstop, the nozzle on my printer is 0.3 mm below the bed so I need to set Z probe offset to +0.3

### Expected behavior

Z probe offset is applied homing with Z min endstop sensor

### Actual behavior

Z probe offset is not applied homing with Z min endstop sensor. It is properly applied only homing with the probe

### Steps to Reproduce

1) configure Marlin to home using Z min endstop (I configured XY outside the bed so the nozzle can reach the Z min endstop without hitting the bed first due to Z min endstop is lower than the bed)
2) set and save a Z probe offset (e.g. +0.3 for my printer)
3) home the printer 
4) raise and move the nozzle on the centre of the bed and lower the nozzle to Z0
5) the nozzle will always scratch/push on the bed. Z0 matches the homing Z0 while I was expecting a +0.3

Tested also using big Z probe offset values (e.g. +30, -5 etc...) but no Z probe offset is applied.
Attached my configurations applied on Marlin bugfix from 2021/08/27.
All the changes are marked with the ending comment "// DIGA-Tech:"

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7070565/Marlin.zip)

### Version of Marlin Firmware

Marlin bugfix from 2021/08/27

DerAndere1, 2021-08-28:
Did you try changing the the `Z_MIN_POS` value ?

thisiskeithb, 2021-08-28:
Duplicate of #21833

closed 2021-08-28 by thisiskeithb

digant73, 2021-08-30:
ok, although it seems to me a different setup.

DerAndere1, 2021-08-31:
Also duplicate of https://github.com/MarlinFirmware/Marlin/issues/21727 . Read the fine manual at https://marlinfw.org/docs/hardware/endstops.html#probe-used-for-homing-and-bed-leveling.

digant73, 2021-08-31:
@DerAndere1 I will try to also use MANUAL_Z_HOME_POS