MarlinFirmware/Marlin #22661: [FR] Platformio env for firmware for Ender 3 PRO into a BIGTREETECH SKR mini E3 v2.0.

gdrDeveloper, 2021-08-29:
### Is your feature request related to a problem? Please describe.

I want to build a firmware for Ender 3 PRO into a BIGTREETECH SKR mini E3 v2.0 and i need envirotment configuration.

I tried to compile with marlin and bugversion 2.0x repositories but both give me the same error

I tried whit this 
--------------------------------------------------------------------------------------------------------------------
#
# Just print the dependency tree
#
[env:STM32F103RC_btt_512K]
platform    = ststm32
board       = genericSTM32F103RC
board_build.core = maple
build_flags = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
src_filter  = +<src/MarlinCore.cpp>
--------------------------------------------------------------------------------------------------------------------

 Error: Build environment 'STM32F103RC_btt_512K' is incompatible with BOARD_BTT_SKR_MINI_E3_V2_0. Use one of these: STM32F103RC_btt, STM32F103RC_btt_USB, STM32F103RE_btt, STM32F103RE_btt_USB, STM32F103RC_btt_maple, STM32F103RC_btt_USB_maple, STM32F103RE_btt_maple, STM32F103RE_btt_USB_maple
--------------------------------------------------------------------------------------------------------------------

I Tried use STM32F103RC_btt too and..

--------------------------------------------------------------------------------------------------------------------
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F103RC_btt\firmware.elf] Error 1
--------------------------------------------------------------------------------------------------------------------

And with STM32F103RC_btt_maple have and error too
--------------------------------------------------------------------------------------------------------------------
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\HAL\shared\eeprom_if_i2c.cpp:28:
c:\marlin-bugfix-2.0.x\marlin\src\hal\stm32f1\inc\sanitycheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
Compiling .pio\build\STM32F103RC_btt_maple\src\src\gcode\control\M211.cpp.o
*** [.pio\build\STM32F103RC_btt_maple\src\src\HAL\shared\eeprom_if_i2c.cpp.o] Error 1
In file included from Marlin\src\feature\../inc/MarlinConfig.h:50:0,
                 from Marlin\src\feature\babystep.cpp:23:
Marlin\src\feature\../inc/../HAL/STM32F1/inc/SanityCheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\inc/MarlinConfig.h:50:0,
                 from Marlin\src\MarlinCore.h:24,
                 from Marlin\src\MarlinCore.cpp:31:
Marlin\src\inc/../HAL/STM32F1/inc/SanityCheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\feature\bedlevel\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\feature\bedlevel\bedlevel.cpp:23:
c:\marlin-bugfix-2.0.x\marlin\src\hal\stm32f1\inc\sanitycheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
In file included from c:\marlin-bugfix-2.0.x\marlin\src\inc\marlinconfig.h:50:0,
                 from Marlin\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp:23:
c:\marlin-bugfix-2.0.x\marlin\src\hal\stm32f1\inc\sanitycheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\feature\../inc/MarlinConfig.h:50:0,
                 from Marlin\src\feature\controllerfan.cpp:23:
Marlin\src\feature\../inc/../HAL/STM32F1/inc/SanityCheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\feature\../inc/MarlinConfig.h:50:0,
                 from Marlin\src\feature\tmc_util.cpp:23:
Marlin\src\feature\../inc/../HAL/STM32F1/inc/SanityCheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
In file included from Marlin\src\feature\../inc/MarlinConfig.h:50:0,
[... 51 lines cut ...]
*** [.pio\build\STM32F103RC_btt_maple\src\src\gcode\config\M301.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_maple\src\src\MarlinCore.cpp.o] Error 1
In file included from Marlin\src\gcode\control\../../inc/MarlinConfig.h:50:0,
                 from Marlin\src\gcode\control\../gcode.h:309,
                 from Marlin\src\gcode\control\M211.cpp:27:
c:\marlin-bugfix-2.0.x\marlin\src\hal\stm32f1\inc\sanitycheck.h:50:4: error: #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
   #error "EMERGENCY_PARSER is only supported by HardwareSerial or USBComposite in HAL/STM32F1."
    ^~~~~
*** [.pio\build\STM32F103RC_btt_maple\src\src\gcode\config\M302.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_maple\src\src\gcode\bedlevel\mbl\G29.cpp.o] Error 1
*** [.pio\build\STM32F103RC_btt_maple\src\src\gcode\control\M211.cpp.o] Error 1
======================================================================================= [FAILED] Took 9.50 seconds =======================================================================================
Environment            Status    Duration
---------------------  --------  ------------
STM32F103RC_btt_maple  FAILED    00:00:09.499
================================================================================= 1 failed, 0 succeeded in 00:00:09.499 ================================================================================= 
The terminal process "~\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

### Are you looking for hardware support?

I want to build a firmware for Ender 3 PRO into a BIGTREETECH SKR mini E3 v2.0.

### Describe the feature you want

Want to clone repository and can build the firmware

ellensp, 2021-08-29:
STM32F103RC_btt_512K no longer exsists. It was a dumb hack that caused more issues that it solved.
As it says Use one of these: STM32F103RC_btt, STM32F103RC_btt_USB, STM32F103RE_btt, STM32F103RE_btt_USB
If you chip says stm32f103rc, use stm32f103rc if your chip says stm32f103re use stm32f103re

closed 2021-08-29 by ellensp

ellensp, 2021-08-29:
Tested ENV:STM32F103RC_btt with https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0
on bugfix It compiles without issues