MarlinFirmware/Marlin #22662: [BUG] Compile Error Z_PROBE_OFFSET_RANGE

sarvenn, 2021-08-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

27th Aug bugfix 2 compile error.
I added the terminal error messeges below. It seems to be due to source code files.
[compile error.txt](https://github.com/MarlinFirmware/Marlin/files/7072292/compile.error.txt)

### Bug Timeline

new

### Expected behavior

Successful compile

### Actual behavior

Unsuccesful compile

### Steps to Reproduce

Using both Autobuild Marlin and Platformio build task have the same error results

### Version of Marlin Firmware

bugfix2 27th Aug

### Printer model

Ender 3 pro

### Electronics

BTT SKR Mini E3 V2 & TFT35 E3 V3

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

ellensp, 2021-08-29:
You need matching Configuration.h files eg it must have this line
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h#L1248

sarvenn, 2021-08-29:
I'm not even using a Z probe. Here are my config files.
[config files.zip](https://github.com/MarlinFirmware/Marlin/files/7072306/config.files.zip)

ellensp, 2021-08-29:
you have
//#define Z_PROBE_OFFSET_RANGE_MIN -20
//#define Z_PROBE_OFFSET_RANGE_MAX 20
You have #define MESH_BED_LEVELING which uses these defines
Enabled these defines.

sarvenn, 2021-08-29:
I also tried for STM32F103RE_btt. Result is the same with STM32F103RC_btt

sarvenn, 2021-08-29:
I think I found the reason of the problem. By default these line were inactive.
I enabled these commands and compiled again. It is successful now.
I think this should be enabled by default.
Actually I use the ender 3 pro skr mini e3 v2 config files and do some minor changes. manual MBL was enabled by default in that config. So Z probe offset range should be enabled by default as well. May be this can be added to sanity check as well. If any of the mesh bed leveling is enabled then z probe offset range also should be enabled. I don't have knowledge about coding. I'm sure you will find the best solution to overcome this error.

// For M851 give a range for adjusting the Z probe offset
#define Z_PROBE_OFFSET_RANGE_MIN -20
#define Z_PROBE_OFFSET_RANGE_MAX 20

ellensp, 2021-08-29:
PR recreated. closing.

closed 2021-08-29 by ellensp