MarlinFirmware/Marlin #22665: [BUG] STM32F103RE_btt with BTT SKR CR6 board: warning: "E2END" redefined

Sebazzz, 2021-08-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling for the BTT SKR CR-6 board, a warning is given:

```
z:\dev\marlin\marlin\src\pins\stm32f1/pins_BTT_SKR_CR6.h:62: warning: "E2END" redefined
   62 | #define E2END           (MARLIN_EEPROM_SIZE - 1)  // 2KB
      |
In file included from C:\Users\sebas\.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:17,
                 from C:\Users\sebas\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
                 from C:\Users\sebas\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:36,
                 from z:\dev\marlin\marlin\src\hal\shared\marduino.h:36,
                 from z:\dev\marlin\marlin\src\hal\stm32\hal.h:28,
                 from z:\dev\marlin\marlin\src\hal\hal.h:30,
                 from z:\dev\marlin\marlin\src\inc\marlinconfig.h:31,
                 from Marlin\src\lcd\dogm\ultralcd_st7920_u8glib_rrd_AVR.cpp:30:
C:\Users\sebas\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/stm32_eeprom.h:83: note: this is the location of the previous definition
   83 | #define E2END (FLASH_PAGE_SIZE - 1)
      |
```

### Bug Timeline

Happens only when compiling for non-maple pio env

### Expected behavior

There should be no warning.

### Actual behavior

There is a warning.

### Steps to Reproduce

1. Use attached config
2. Compile with pio env STM32F103RE_btt. Does not happen with `_maple`.
3. Warning galore

### Version of Marlin Firmware

Latest

### Printer model

Creality CR-6 SE

### Electronics

BTT SKR CR6 board; filament sensor (stock)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[btt-skr-cr6-with-btt-tft-e3d-hemera.zip](https://github.com/MarlinFirmware/Marlin/files/7073027/btt-skr-cr6-with-btt-tft-e3d-hemera.zip)

ellensp, 2021-08-30:
This silences it.  Update Marlin/src/pins/stm32f1/pins_BTT_SKR_CR6.h
Change
```CPP
#define E2END           (MARLIN_EEPROM_SIZE - 1)  // 2KB
```
to
```CPP
#ifdef __STM32F1__
  #define E2END           (MARLIN_EEPROM_SIZE - 1)  // 2KB
#endif
```

thinkyhead, 2021-08-31:
`E2END` does not need to be defined by Marlin. However, Marlin uses `E2END` when `MARLIN_EEPROM_SIZE` has not been provided by the pins file. So you should just remove that obsolete line from the pins file.

commit 2c5daa07 (2021-08-31): 🩹 Clean up BTT_SKR_CR6

Fixes #22665
closed 2021-08-31 by thinkyhead