MarlinFirmware/Marlin #22666: [BUG] BTT SKR CR-6 with STM32F103RE_btt: With SDSUPPORT bootloop (watchdog) / freeze (no watchdog)

Sebazzz, 2021-08-29:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Seems similar to #22078 where the user tracked the issue to `SDSUPPORT`. That was an SKR board, this is an SKR board. I have reproduced this on latest bugfix-2.0.x. This issue does not happen on maple builds.

Booting:
- with SD inserted: immediate boot loops without user interaction
- without SD inserted: stable, until SD is inserted and then the behavior in "with SD inserted" happens

I haven't tried whether the other issues exist, but I assume the software serial also exists because this board also has software serial.

### Bug Timeline

Happens only when compiling for non-maple pio env

### Expected behavior

I expected the printer to be able to boot and use the SD card.

### Actual behavior

The printer cannot be used with an SD card. Reboots or freezes depending on USE_WATCHDOG.

### Steps to Reproduce

1. Use attached configs
2. Compile with `STM32F103RE_btt`
3. 

### Version of Marlin Firmware

Latest

### Printer model

Creality CR-6 SE

### Electronics

BTT SKR CR6 board; filament sensor (stock)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[btt-skr-cr6-with-btt-tft-e3d-hemera.zip](https://github.com/MarlinFirmware/Marlin/files/7073052/btt-skr-cr6-with-btt-tft-e3d-hemera.zip)

Sebazzz, 2021-08-29:
FYI @rhapsodyv since you were involved with #22078

Sebazzz, 2021-08-29:
Opened #22667 which does seem to resolve the SD card issue, however, won't this issue be present on more SKR boards?

Sebazzz, 2021-08-29:
Serial issue also appears to be here.

thisiskeithb, 2021-08-29:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/22668

closed 2021-08-29 by thisiskeithb