MarlinFirmware/Marlin #22670: Maple Fails

khitengoku, 2021-08-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

maple builds for skr mini e3 boards for ender3  (STM32F103RC_btt_maple) does not work in the 2.9 versions.  Worked fine in 2.8.  However, in 2.9 it freezes at the Marlin boot screen.  I tried both stable and bugfix.  Without this build, marlin will NOT communicate with pronterface/printrun via USB.    

### Bug Timeline

2.9

### Expected behavior

Marlin to load 

### Actual behavior

Doesn't load

### Steps to Reproduce

Build a maple on the STM32 environment for BTT.  

### Version of Marlin Firmware

2.9

### Printer model

Ender 3

### Electronics

SKR mini e3 v 2.0

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/7073907/Archive.zip)
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/7073912/Archive.zip)

One of these is from stable build the other is from bugfix if i picked the correct files.

thinkyhead, 2021-08-31:
I'm certainly not the expert around here on the subject of BigTreeTech builds.

- What do you suppose was changed in the Maple build between 2.0.8 and 2.0.9.x that causes it to freeze?
- Why do you suppose that the HAL/STM32 build now has a broken USB connection? Has it always been thus?

CC: @bigtreetech

khitengoku, 2021-08-31:
> 
> 
> I'm certainly not the expert around here on the subject of BigTreeTech builds.
> 
> What do you suppose was changed in the Maple build between 2.0.8 and 2.0.9.x that causes it to freeze?
I don’t know for sure.  I compared .h and .adv between 2.9 and working 2.8.  No real changes.   The standard build works fine except no communication to Pronterface or cura.   That holds true even in 2.8.  For comms in 2.8 you have to use maple build.   

> Why do you suppose that the HAL/STM32 build now has a broken USB connection? Has it always been thus?
I have used the skr mini e3 v1.2 for a year with no issue.  Over the last month I’ve been helping people with their 2.0. So I just became aware of an issue.  So I purchased a 2.0 to test.   Same issues they had.   When you use 2.9 and a standard build the printer functions as expected except no serial communication on either windows or Mac OS environment.   When you use 2.8 standard build the results are the same.  When you build using the 2.8 maple you now get full functionality.   2.7 didn’t have a maple that I can see.   When you try 2.9 maple you freeze at boot screen when first loading the bin file.   Even a hard reset does nothing.   

All in all, the issue lies somewhere in how 2.9 us utilizing the maple environment.  Something changed between 2.8 and 2.9.  But to be honest, I haven’t taken the time to delve fully into all the sub files to see where that change may be.

tpruvot, 2021-08-31:
try to unplug the sdcard, and to format it (if that fixes the boot freeze)

khitengoku, 2021-08-31:
I tried 3 different known ender cards.   However, I complied with your request.  A newly formatted as card (fat32) and maple build still freezes at marlin load screen.   Standard build still loads fine.

khitengoku, 2021-08-31:
I want to clarify.  The problem is in marlin.  Not the hardware.  I have three ender printers side by side.  I have tested 2.0.8 and 2.0.9.  I have built a STM32F103RC_btt for both and a ..... btt_maple for both.   The 2.0.9 maple fails to load.  All other builds load fine.   The sir mini e3 v2 requires the maple build to communicate via usb serial with Pronterface.  I don’t have an issue using 2.0.8 (partly because I’m not a fan of some of the menu changes made in 2.0.9).  However, if this problem becomes persistent, it limits the future of my marlin with an skr mini e3 v2 board.

tpruvot, 2021-08-31:
thats weird, maple/F1 HAL didnt really change recently, but i may have to check the logs to be sure

tpruvot, 2021-08-31:
could you rename the issue title, its not a "build issue", more a "freeze/boot" issue

khitengoku, 2021-08-31:
I will attempt to.   But this freeze boot is caused by something changed in the marlin 2.0.9 build.  As I said, it does not exist in the 2.0.8.  And it is specifically tied to the maple version of the ST IN the 2.0.9 build environment.    It exists in both the “stable” and the bugfix of 2.0.9.  So I guess my question is “why does it only affect the maple build, which is necessary for the skr mini e3 v2?”

tpruvot, 2021-08-31:
its required by all F1 devices, STM32 portage is still experimental/under work

khitengoku, 2021-08-31:
Id like to share the known working build, btt's 2.8.2, but it keeps telling me "not included in the list"

tpruvot, 2021-08-31:
2.0.8.2 ?

khitengoku, 2021-08-31:
yea.  but it is one i got directly from bigtreetech.  I didn't remember that part.  I went back and downloaded your direct 2.0.8.2 and it doesn't have the same maple build environment.  This is the btt version screen. 
[image]

thinkyhead, 2021-09-04:
FWIW, note that the environments we now label with "maple" were (in many cases) the _only_ build environments in previous versions of Marlin. We added new environments that build with HAL/STM32 and renamed the old ones with a "`_maple`" suffix.

thinkyhead, 2021-09-04:
ICYMI — Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

khitengoku, 2021-09-05:
I used the link you sent, I grabbed the bugfix config files.  I set the parameters for my bltouch.  The STM32F103RC_BTT_MAPLE still failed to load (freezes at the load screen where it says Marlin and the firmware).  Then i tried the standard 103RC_BTT.  it loads but does not have usb serial ability.  Then I tested the 103RC_BTT_USB. still no usb serial interface.

khitengoku, 2021-09-05:
By the way, here is the  bugfix i coded, ifyou need it. https://drive.google.com/file/d/1M9-a47LxYZR8Z6iJIO8dHc0lo23coAu0/view?usp=sharing

Z18Aplha, 2021-12-25:
Hey,

i experience the same issue. I wanted to update my old Marlin version to the recent one. But unfortunately the firmware upgrade with the sd card doesnt work. 

When powering on the printer with the sd card plugged in (only the fresh compiled firmware.bin) the screen stays blue (forever). Not even the Marlin logo shows up. It just freezes. A reboot doesnt changes something but the file on the sd card is renamed to FIRMWARE.CUR. So something happened but didn't worked.

With the firmware from Bigtreetech (also mentioned in this thread already) which is based on 2.0.8 the firmware upgrade works as expected. I have no idea why, but something seems to be broken with the 2.0.9 and the skr mini e3 v2 board.

khitengoku, 2021-12-25:
As far as I know, there was never a solution provided.

suleytatar, 2021-12-25:
I got the same issue, bootload from SD, failure on MKS_SBASE, with 2.0.9.3. My configs was working fine with 2.0.9.2. compiled new release today with platformio, new build firmware.bin is read by the board(MKS_SBASE) ::> led get stuck, only 4th is lit, and LCD is frozen, select button does not work . Reset  reboots to same state. 
what would the next step to fix would be??

Rincewind08, 2022-01-22:
As above described, my gtr no longer works as of this last two updates of marlin.BTT  GTR v1.0 
I do not understand why, but neither the bugfix nor the regular vrsion of marlin will hget past the boot screen, though the firmware is always renamed to FIRMWARE.CUR, and the firmware.bin is always removed from sd card.  I have tried 9 seperate sd cards, each with a fresh format.  Rolling back to previous marlin builds fixes the problem.

suleytatar, 2022-01-22:
The issue I had; which is similar to described above; has been cleared after a new/fresh clone of bugfix2.0.x, with imported and edited old config files.

waynert3d, 2022-04-08:
I would say I have the saWont go past Marlin logo. 2.0.9.2 was working fine and Decided to go to 2.0.9.3. It is on a chitu version six board on 330x330x400 tronxy. Compiling useing platformio with no errors. Other than dimesions, probe offset, and temperatures set, no speaker all other settings left as is. Cannot find the problem. New firmware is loaded with screen black , then flashes white and then the logo appears and it now stays on logo. If I can be of help please tell me what to do

closed 2022-06-18 by github-actions[bot]