MarlinFirmware/Marlin #22674: 'X_MIN' was not declared in this scope; did you mean 'Z_MIN'?

Deema35, 2021-08-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

When I compile Marlin Bugfix 2.0.x. I have got error:

Marlin\src\module/endstops.h:69:46: error: 'X_MIN' was not declared in this scope; did you mean 'Z_MIN'?

I dont have x min endstop, because I use X max endstop this is my settings;
//#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_IMIN_PLUG
//#define USE_JMIN_PLUG
//#define USE_KMIN_PLUG
#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
//#define USE_IMAX_PLUG
//#define USE_JMAX_PLUG
//#define USE_KMAX_PLUG

### Bug Timeline

Just now

### Version of Marlin Firmware

02000901

tpruvot, 2021-08-30:
in Marlin/Configuration.h set #define X_HOME_DIR 1 instead of -1

Deema35, 2021-08-30:
Thank you, it's working.

closed 2021-08-30 by Deema35