MarlinFirmware/Marlin #22675: Not working LCD display.

Deema35, 2021-08-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

HI. I use BTT_SKR_MINI_V1_1 with LCD 128x64 (connected by I2C interface to board) with encoder. In Marlin I chouse options #define OLED_PANEL_TINYBOY2 and it work, on previous build ( about 1/2 years ago). But now if I do same it don't work  only black screen (not work in 2.0.x not work in bugfix-2.0.x.). I had noticed new file firmware.bin has only 182kB, although the previous one was 194kB. I use same Configurations. On build which work version "2.0.8.2" STRING_DISTRIBUTION_DATE "2021-05-29".

### Bug Timeline

Just now.

### Expected behavior

Screen will work.

### Actual behavior

Screen not work.

### Version of Marlin Firmware

bugfix-2.0.x

ellensp, 2021-08-30:
This board has 4 possible build environments.
env:STM32F103RC_btt env:STM32F103RC_btt_USB env:STM32F103RC_btt_maple env:STM32F103RC_btt_USB_maple
Did you try both the none maple and maple versions?

Deema35, 2021-08-30:
I always use STM32F103RC_btt_USB and it worked. But now work only STM32F103RC_btt_maple and STM32F103RC_btt_USB_maple.

ellensp, 2021-08-31:
That mean there is probably an issue in the non maple env's for I2C.  
This should be left open to be fixed, as the maple env's are end of life and will be removed one day.

thinkyhead, 2021-08-31:
Sorry about the regression. There have been various changes in build environments, pins, and LCD flags, and the STM32 platform itself is also being updated semi-regularly. Maybe this is something which would be fixed by going back to STSTM32@~11.0 or some other previous version, or maybe it has to do with changes to build flags. There is no straightforward way to isolate the issue without direct testing.

Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The first step is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

tpruvot, 2021-08-31:
STM32F103RC_btt_USB was renamed to STM32F103RC_btt_USB_maple, like all STM32F1 devices. The non-maple envs use a more recent stm32duino framework common to all STM chips and may need serious real tests to find differences or regressions. Its not really a recent change in the framework, its a massive change for maintainers.

closed 2022-01-06 by github-actions[bot]