MarlinFirmware/Marlin #22676: [BUG] Extruder stops randomly during prints

luizmont, 2021-08-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I was running a previous version of the bugfix, which I don' recall exactly when it was compiled (I formatted my PC...), but I believe it was around november/2020.
Printer was working without any problems.

I compiled the latest bugfix on this weekend and this is where things start to get weird.
The extruder motor stops working in the middle of the print.
I tried different models and slicers.
My main slicer is Cura, but I tried with Prusa Slicer and the baseline cube from Teaching Tech github page, all with the same results.
I managed to complete the cube once, but when I tried it again, it failed almost at the end of the print (it started printing the Z letter).
It isn't always in the same spot that the extruder stops, it is random.

I'm running a modified Ender 3 Pro, with microswiss direct drive all metal, bltouch, smart filament sensor, TFT35 display and SKR Mini E3 v2 board.
The motor isn't hot to touch, the drivers have heatsinks installed.

Once the problem starts, even if I stop the print the motor won't come back on, I have to power off the printer.

It isn't a clogging issue, because I can extrude manually when the fail happens and the filament extrudes normally.

I tried printing through octoprint and the onboard SD, all with the same results.

Linear advance is enabled on firmware. It fails with K = 0.12, but it also fails with K = 0.

Browsing the open issues I managed to find this one #21390 that seems pretty similar to my situation.

### Bug Timeline

Started after installation of the current bugfix version

### Expected behavior

Finish printing

### Actual behavior

Extruder motor stops in the middle of the print

### Steps to Reproduce

1. Slice model
2. Start printing
3. Fail occurs

### Version of Marlin Firmware

Marlin bugfix-2.0.x

### Printer model

Creality Ender 3 Pro

### Electronics

SKR Mini E3 v2

### Add-ons

bltouch, microswiss direct drive, smart filament sensor

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

To setup the firmware, I used the example from the folder, and aditional configuration from [https://www.reddit.com/r/3Dprinting/comments/n7or5u/marlin_20x_guide_skr_mini_e3_v20_ender_3/](url)

thisiskeithb, 2021-08-30:
Duplicate of #21390

closed 2021-08-30 by thisiskeithb

jeremypatterson, 2021-09-18:
@luizmont I don't know if this helps, but I have a nearly identical setup to yours. It turns out my problem was that the motor extension cable that came with my Micro Swiss unit was defective. I replaced almost all my parts, and burned through a ton of firmware builds before finding this out.

a0s, 2021-09-19:
It seems i've got the same problem with enabled:
```C
#define LIN_ADVANCE
#define LIN_ADVANCE_K 0
```
at 10fda222ea40af91ff1244ad5b7c8737043e1011

5ft24dave, 2021-09-20:
> @luizmont I don't know if this helps, but I have a nearly identical setup to yours. It turns out my problem was that the motor extension cable that came with my Micro Swiss unit was defective. I replaced almost all my parts, and burned through a ton of firmware builds before finding this out.

I have the Micro-Swiss direct drive kit as well.  Worked fine with the creality 4.2.7 board until I loaded Marlin 2.0.9.1  All the test gcode from the teachingtech website stops extruding after about 10-20mm.  doesn't restart for the next layer etc.  If I slice any STL of my own with Cura, it prints with no issues at all  the web site says all their test gcode is sliced using simplify3D.  all of the test gcode files print fine on my Anet A8, and also printed fine on the Ender 3 with the ancient stock motherboard running Marlin 1.1.9.1  One thing I am noticing is there is no feedrate after the Extruder distance commands in the gcode.  Does Marlin 2.0.x need explicit feed rates on each command instead of a global feed rate on the first one to carry it over?

luizmont, 2021-09-20:
> @luizmont I don't know if this helps, but I have a nearly identical setup to yours. It turns out my problem was that the motor extension cable that came with my Micro Swiss unit was defective. I replaced almost all my parts, and burned through a ton of firmware builds before finding this out.

Absolutely! It ended being a problem on the cable.
I even purchased a new stepper motor... 
I used the new cable and the problem went away.

@5ft24dave Have you checked your cables?
It was my problem and jeremypatterson as well!

5ft24dave, 2021-09-20:
Verified cable was good, so decided to start with the basic config and no changes.  Worked fine.  Slowly added features until it stopped.  Linear advance being enabled is what kills it.  Going to try 2.0.8.2 and see if it is still an issue there.

5ft24dave, 2021-09-20:
same under 2.0.8.2  enabling linear advance and it dies.

5ft24dave, 2021-09-20:
found the issue... its listed as this issue.  Stealthchop mode on the TMC22xx drivers causes it.
https://github.com/MarlinFirmware/Marlin/issues/16076