MarlinFirmware/Marlin #22677: Trigorilla Pro has no display

rq3, 2021-08-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Downloaded the latest bugfix, and hand edited the pins and environment per the latest "fix". Marlin compiles OK into a 216KB firmware.bin (which is unusually small).

Installed on printer. No display or touchscreen activity at all.

### Bug Timeline

Old issue. 2.0.9 works well. All releases after that do not.

### Expected behavior

I expected the newly compiled bugfix to work.

### Actual behavior

The printer has no display activity at all. Blank screen.

### Steps to Reproduce

1) Download the latest bugfix (8/30/2021) and its configuration files for the Anycubic Predator delta printer.
2) Compile under PlatformIO.
3) Flash new firmware.bin to printer.
4) Turn printer on and witness no display.

### Version of Marlin Firmware

latest bugfix

### Printer model

Anycubic Predator with stock Trigorilla Pro board and touchscreen

### Electronics

stock

### Add-ons

none

tpruvot, 2021-08-30:
you already submitted issue #22606

thisiskeithb, 2021-08-30:
Duplicate of #22606

closed 2021-08-30 by thisiskeithb

tpruvot, 2021-08-30:
There is a PR #22610 open also...