MarlinFirmware/Marlin #22679: Freeze (dark screen) after sending empty G-code from G-code menu (TS35 + MKS Robin Nano V3 screen)

1qaz2004, 2021-08-30:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

No ignoring empty G-code line

### Bug Timeline

Detected when trying to exit the G-code menu

### Expected behavior

Exit to preview or main screen is pending

### Actual behavior

Dark screen without the possibility of further work

### Steps to Reproduce

Restart firmware or shutdown

### Version of Marlin Firmware

Marlin-bugfix-2.0.x

### Printer model

ANET E16

### Electronics

TS35 + MKS Robin Nano V3 screen

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

[Files.ZIP](https://github.com/MarlinFirmware/Marlin/files/7079524/Files.ZIP)

thinkyhead, 2021-08-31:
Did you actually test with `bugfix-2.0.x`? You provided configurations for Marlin 2.0.8 instead of configurations that are compatible with `bugfix-2.0.x`. Please post updated configurations so that we can actually compile and test.

1qaz2004, 2021-08-31:
Configuration (adv) .h files updated

ellensp, 2021-09-01:
Can you elaborate on
Steps to Reproduce: Restart firmware or shutdown
It doesn't seem related to "Freeze sending empty G-code from MKS TFT UI "

1qaz2004, 2021-09-01:
It has been empirically found that the printer remains in working order, since it is controlled via USB via Repetier. Screen health is restored after the G28 from Repetier. Perhaps Marlene is just waiting for the G-code command. But a dark (blank) screen is not the normal state of processing an empty line.

thinkyhead, 2021-09-07:
String handling in Marlin is pretty primitive and lacks bounds-checking to make it more robust. That will be overhauled at some point. Meanwhile, it's quite possible that some buffer is overflowing or otherwise improperly handled in the screen code that was contributed by Makerbase. I'm not familiar with the code in question so it will take some time to analyze.

thinkyhead, 2021-09-08:
Let's see if this helps. A simple change in `draw_keyboard.cpp` (line 165)…
```diff
-         if (!queue.ring_buffer.full(3)) {
+         if (strlen(ret_ta_txt) && !queue.ring_buffer.full(3)) {
```

1qaz2004, 2021-09-09:
Hello, unfortunately the proposed changes in line 165 did not fix the problem. It may be necessary to modify the file Marlin-bugfix-2.0.9.1 \ Marlin \ src \ lcd \ extui \ mks_ui \ draw_gcode.cpp
Thanks.

1qaz2004, 2021-10-20:
Sending empty G-code from MKS interface resolved. Now processing an empty command does not cause a black screen. In addition, I would like to note that for successful compilation I had to make changes to the file
Marlin-bugfix-2.0.x \ Marlin \ src \ pins \ stm32f4 \ pins_MKS_ROBIN_NANO_V3_common.h

-     169    #undef PSU_CONTROL
       170    #undef MKS_PWC
       171    #define SUICIDE_PIN                          PB2
       172    #define SUICIDE_PIN_STATE               LOW

+    169    //#undef PSU_CONTROL
       170    #undef MKS_PWC
       171    #define PS_ON_PIN                          PB2  // SUICIDE_PIN
       172    #define PS_ON_PIN _STATE              LOW // SUICIDE_PIN_STATE