MarlinFirmware/Marlin #22681: EVENT_GCODE_TOOLCHANGE_[BUG] doesn't work (bug summary)

xzy95024, 2021-08-31:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to add a fifth axis based on the 2.0.9 Marlin. It works well somehow（the pronterface can control the fifth axis). Now I am trying to use EVENT_GCODE_TOOLCHANGE_T0 to run the fifth axis. However, nothing happened. Here is my code:#define EVENT_GCODE_TOOLCHANGE_T0 "G1 A10" .

Beside that I found if I send the same G code twice（Like G1 A10）,the second one won't work. And the motor direction is random. Here are my configuration.

### Version of Marlin Firmware

2.0.9

### Printer model

customized

### Electronics

board:mks robin pro

### Host Software

Pronterface

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/7082569/Marlin.zip)

DerAndere1, 2021-08-31:
Please test the following amd describe observations for all steps that proceed different from the expectation:

1. flash the firmware
2. Send `M502`
3. Send `M500`
4. Run the following Gcode:
```
G28 ; homing
G90 ; relative mode
G1 A10 ; Should move 10 units in positive direction
G1 A10 ; Should move 10 units in positive direction 
T1 ; Observe toolchange to tool 1
T0 ; toolchange to tool 0 accompanied by a move of the A stepper for 10 units in positive direction.
T1 ; Observe toolchange to tool 1
T0 ; toolchange to tool 0 accompanied by a move of the A stepper for 10 units in positive direction.
G1 A10 ; Should move 10 units in positive direction 
G90 ; Absolute mode
G1 A10 ; Should move to absolute position A10
G1 A10 ; Should stay at absolute position A10
```

xzy95024, 2021-09-01:
> Please test the following amd describe observations for all steps that proceed different from the expectation:
> 
> 1. flash the firmware
> 2. Send `M502`
> 3. Send `M500`
> 4. Run the following Gcode:
> 
> ```
> G28 ; homing
> G91 ; relative mode
> G1 A10 ; Should move 10 units in positive direction
> G1 A10 ; Should move 10 units in positive direction 
> T1 ; Observe toolchange to tool 1
> T0 ; toolchange to tool 0 accompanied by a move of the A stepper for 10 units in positive direction.
> T1 ; Observe toolchange to tool 1
> T0 ; toolchange to tool 0 accompanied by a move of the A stepper for 10 units in positive direction.
> G1 A10 ; Should move 10 units in positive direction 
> G90 ; Absolute mode
> G1 A10 ; Should move to absolute position A10
> G1 A10 ; Should stay at absolute position A10
> ```

Thanks for reply

G91; Seems good
G1 A10: It works
G1 A10: nothing happened
T1:nothing happened
T0:Nothing happened
T1:nothing happened
T0:Nothing happened
G1 A10:Nothing happened
G90： not sure whether it works or not
G1 A10：Nothing happened
G1 A10: Nothing happened
G1 A20: work （clockwise）
G1 A10: work （conterclockwise）
G1 A10: nothing happened

ellensp, 2021-09-01:
From those results it would seem that Axis A is stuck in absolute mode....

ellensp, 2021-09-01:
Using your config I see a bunch of nasty looking warnings when compiling. (array overruns)
Seems to be caused by the following in  Marlin/src/module/motion.cpp
`    HOTEND_LOOP() LOOP_LINEAR_AXES(a) hotend_offset[e][a] = tmp[a][e];`
To me this should be 
`    HOTEND_LOOP() LOOP_ABC(a) hotend_offset[e][a] = tmp[a][e];`

Could you update this line, Perhaps this is causing the issue.

DerAndere1, 2021-09-01:
maybe it is better to do the following changes (I don't think they will fix the issue, though).
In Configuration.h:
```
#if LINEAR_AXES >= 4
  #define HOTEND_OFFSET_I { 0.0, 0.0 }
#endif
#if LINEAR_AXES >= 5
  #define HOTEND_OFFSET_J { 0.0, 0.0 }
#endif
#if LINEAR_AXES >= 6
  #define HOTEND_OFFSET_K { 0.0, 0.0 }
#endif
```

motion.hcpp L121:
```
constexpr float tmp[LINEAR_AXES][HOTENDS] = LINEAR_AXIS_ARRAY(HOTEND_OFFSET_X, HOTEND_OFFSET_Y, HOTEND_OFFSET_Z, HOTEND_OFFSET_I, HOTEND_OFFSET_J, HOTEND_OFFSET_K);
```

motion.cpp L127:
```
HOTEND_LOOP() LOOP_LINEAR_AXES(a) hotend_offset[e][a] = tmp[a][e];
```

If above changes did not fix this issue, try to change
https://github.com/MarlinFirmware/Marlin/blob/01d1192a441f5afb753185eb103be64baa81bd43/Marlin/src/core/types.h#L75-L111
to the following:
```
enum AxisEnum : uint8_t {
  X_AXIS = 0, A_AXIS = X_AXIS,
  Y_AXIS = 1, B_AXIS = Y_AXIS,
  Z_AXIS = 2, C_AXIS = Z_AXIS,
  #if LINEAR_AXES >= 4
    I_AXIS,
  #endif
  #if LINEAR_AXES >= 5
    J_AXIS,
  #endif
  #if LINEAR_AXES >= 6
    K_AXIS,
  #endif
  E_AXIS,
  X_HEAD, Y_HEAD, Z_HEAD,
  E0_AXIS = E_AXIS,
  E1_AXIS, E2_AXIS, E3_AXIS, E4_AXIS, E5_AXIS, E6_AXIS, E7_AXIS,
  ALL_AXES_ENUM = 0xFE, NO_AXIS_ENUM = 0xFF
};

//typedef IF<(NUM_AXIS_ENUMS > 8), uint16_t, uint8_t>::type axis_bits_t;
```

xzy95024, 2021-09-02:
> Using your config I see a bunch of nasty looking warnings when compiling. (array overruns)
> Seems to be caused by the following in Marlin/src/module/motion.cpp
> ` HOTEND_LOOP() LOOP_LINEAR_AXES(a) [...]
It's great to see your reply on a working day morning! (different time zones), Your code is work! Now I can switch the motor to relative mode!  Thanks a lot!

What confuses me is that even it is in the relative mode, the event g code tool-change doesn't work.

xzy95024, 2021-09-02:
> Please test the following amd describe observations for all steps that proceed different from the expectation:
>
> 1. flash the firmware
> 2. Send `M502`
> 3. Send `M500`
> 4. Run the following Gcode:
>
> ``` [...]

> maybe it is better to do the following changes (I don't think they will fix the issue, though).
> In Configuration.h:
>
> ```
> #if LINEAR_AXES >= 4
>   #define HOTEND_OFFSET_I { 0.0, 0.0 }
> #endif
> #if [...]

Hey~ Thanks for your reply. I updated the code "HOTEND_LOOP() LOOP_ABC(a) hotend_offset[e][a] = tmp[a][e];"
 Now it seems can work in both relative and absolute mode. The only question is the event_gcode_toolchange still doesn't work. Any suggestions? Again, thanks a lot for your reply

DerAndere1, 2021-09-02:
To confirm a bug in the AxisEnum, can you please change the code in types.h L75 ff back to
```
 enum AxisEnum : uint8_t { 
  
   // Linear axes may be controlled directly or indirectly 
   LINEAR_AXIS_LIST(X_AXIS, Y_AXIS, Z_AXIS, I_AXIS, J_AXIS, K_AXIS) 
  
   // Extruder axes may be considered distinctly 
   #define _EN_ITEM(N) , E##N##_AXIS 
   REPEAT(EXTRUDERS, _EN_ITEM) 
   #undef _EN_ITEM 
  
   // Core also keeps toolhead directions 
   #if EITHER(IS_CORE, MARKFORGED_XY) 
     , X_HEAD, Y_HEAD, Z_HEAD 
   #endif 
  
   // Distinct axes, including all E and Core 
   , NUM_AXIS_ENUMS 
  
   // Most of the time we refer only to the single E_AXIS 
   #if HAS_EXTRUDERS 
     , E_AXIS = E0_AXIS 
   #endif 
  
   // A, B, and C are for DELTA, SCARA, etc. 
   , A_AXIS = X_AXIS 
   #if LINEAR_AXES >= 2 
     , B_AXIS = Y_AXIS 
   #endif 
   #if LINEAR_AXES >= 3 
     , C_AXIS = Z_AXIS 
   #endif 
  
   // To refer to all or none 
   , ALL_AXES_ENUM = 0xFE, NO_AXIS_ENUM = 0xFF 
 }; 
  
 typedef IF<(NUM_AXIS_ENUMS > 8), uint16_t, uint8_t>::type axis_bits_t; 
```
Then run the following test:
1. flash
2. `M502`
3. `M500`
4. run the following gcode and report what steps of the following test fail?:
```
G28 ; homing
G91 ; relative mode
G1 A10 ; Should move 10 units in positive direction
G1 A10 ; Should move another 10 units in positive direction 
G1 A-10 ; Should move 10 units back 
G90 ; absolute mode
G1 A15 ; Should move to absolute position A15
G1 A15 ; Should stay at absolute position A15
```

DerAndere1, 2021-09-02:
I could not reproduce the issue with switching between absolute / relative mode so far, for me it works even with `EXTRUDERS 2`. but I could reproduce the non-working `EVENT_GCODE_TOOLCHANGE_T1` and `EVENT_GCODE_TOOLCHANGE_T1_T0`. For me, this issue occurs only if the machine is in a state where one ore more axes need homing. Lesson learned:  Always have G28 in the beginning of your gcode script or in the start-gcode. 

As a side note: it may be unsafe to rely on relative mode, especially with `EVENT_GCODE_TOOLCHANGE_T1` and `EVENT_GCODE_TOOLCHANGE_T0` . You may want to explicitly change to absolute mode during the toolchange, especially if the A axis is used to drive the tool-changer. Suppose the tool changer should always move to absolute position A10 for tool 0 and to absolute position A0 for tool 1, then better use
`#define EVENT_GCODE_TOOLCHANGE_T0 "G90\nA10"` 
`#define EVENT_GCODE_TOOLCHANGE_T1 "G90\nA0"`

DerAndere1, 2021-09-02:
If for a reason you must leave an axis unhomed and still want to execute the event gcode, then you may try the following:
https://github.com/MarlinFirmware/Marlin/blob/718227a94c0cb163a73f0f288be6f7b864b7127a/Marlin/src/module/tool_change.cpp#L1316-L1331

Change that to
```
    if (!no_move) {
      #ifdef EVENT_GCODE_AFTER_TOOLCHANGE
        if (TERN1(DUAL_X_CARRIAGE, dual_x_carriage_mode == DXC_AUTO_PARK_MODE))
          gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_AFTER_TOOLCHANGE));
      #endif
    }

    #ifdef EVENT_GCODE_TOOLCHANGE_T0
      if (new_tool == 0)
        gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_TOOLCHANGE_T0));
      #endif

    #ifdef EVENT_GCODE_TOOLCHANGE_T1
      if (new_tool == 1)
        gcode.process_subcommands_now_P(PSTR(EVENT_GCODE_TOOLCHANGE_T1));
    #endif
```

DerAndere1, 2021-09-07:
Also (probably uneated:
https://github.com/MarlinFirmware/Marlin/blob/0f61d9e4dd4d4e4f27e5c688ab2c5dbd0f03af84/Marlin/src/module/stepper/trinamic.cpp#L845-L856
change that to 
  #if AXIS_IS_TMC(X)
    TMC_INIT(X, STEALTH_AXIS_XY);
  #endif
  #if AXIS_IS_TMC(X2)
    TMC_INIT(X2, STEALTH_AXIS_XY);
  #endif
  #if AXIS_IS_TMC(Y)
    TMC_INIT(Y, STEALTH_AXIS_XY);
  #endif
  #if AXIS_IS_TMC(Y2)
    TMC_INIT(Y2, STEALTH_AXIS_XY);
  #endif

DerAndere1, 2021-09-23:
thinkyhead added the `HOTEND_LOOP() LOOP_ABC(a) hotend_offset[e][a] = tmp[a][e];` fix by @ellensp to bugfix-2.0.x. Can this issue be closed?

thisiskeithb, 2021-09-23:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/35ad3b0d3b1a8bbf6b220f30f63fd0e32be8fe34

closed 2021-09-23 by thisiskeithb