MarlinFirmware/Marlin #22684: [FR] I want to choose the number of mesh points on G29 command

aalku, 2021-08-31:
### Is your feature request related to a problem? Please describe.

I am using bilinear bed leveling with 10x10 points but sometimes I print on top of another existing object so I glue it to the center of the bed and level that zone with G29 Lx Rx Fx Bx but it uses 10x10 points because that's hardcoded in the firmware and that is soo sloooow. RepRap's G29 allows a Px parameter with the number of points. I wish to have that in Marlin.

### Describe the feature you want

RepRap's G29 allows a Px parameter with the number of points. I wish to have that in Marlin. The number of points would be a couple of extra bytes in the mesh structure rather than a pair of hardcoded constants. The hardcoded value would be the default.

### Additional context

I think I can code it myself but I wish to have some guidance so the change is accepted. I don't want to mantain a fork.

mydevpeeps, 2021-09-20:
First I want to say, G29.cpp shows P is available for LINEAR leveling only.  Why wouldn't we want to use this with BILINEAR?? I await an answer on that one for sure.

``

I had some thoughts on this one and this helps me solidify them. I have pondered a way to speed up G29. It does have all kinds of settings to mesh a specific area of the bed.. but as you pointed out.. if it is doing 10 points.. then it does 10 points within that space. There is also the concept of a rectangular bed shape, which then makes your 10x10 grid not uniform in each axis. If you have a 100mm (Y) x 200mm (X) bed then you should have double the probe points on the X axis than the Y axis. You can set these different in the firmware and using X and Y as params, but again this is only configured for LINEAR leveling.  In reference to just the P parameter on G29 you can't set them separately.

My idea is to make it automatic. Initially this could be done in configuration at compile time, but I'd like to see it happen automatically when G29 runs.

My suggestion is to instead introduce a constant and some logic in Configuration.h that defines the # of probe points/ x mm. 

Here would be an example in code:
`#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
 
  // Calculate the # of grid points based on bed size (offset is factored in later)
  // X-Axis Example: 200mm o X / 50mm grid distance = 5 points (0,50,100,150,200)
  // Y-Axis Example: 100mm o Y / 50mm grid distance = 3 points (0,50,100)

  #define CALCULATE_GRID_MAX_POINTS
  #if ENABLED(CALCULATE_GRID_MAX_POINTS)  
    #define G29_GRID_DISTANCE 50    // distance in mm between probe points
    #define GRID_MAX_POINTS_X (( X_BED_SIZE / G29_GRID_DISTANCE ) + 1 )
    #define GRID_MAX_POINTS_Y (( Y_BED_SIZE / G29_GRID_DISTANCE ) + 1 )
  #else
    #define GRID_MAX_POINTS_X 5     // manually set grid points for x plane
    #define GRID_MAX_POINTS_Y 5     // manually set grid points for y plane
  #endif
`
Doing it this way means that when G29 runs, it's using a distance value instead of a hard coded _number of points_ value. How is this helpful? If you still recognize the other parameters on the G29 command, it will see that your defined "size" of the mesh request is 100mmx100mm instead of the bed size of 100mmx200mm. With a distance of 50mm set, this gives you a grid of points that is 3x3 (9) instead of 5x3 (15). In my opinion this is better than the G29 P parameter which binds a value to both axis.

This could be added right away in Configuration.h to put some sanity to a default value.  G29 would use this value as it does today. If you wanted to take this a step further, you could code G29.cpp to just use the G29_GRID_DISTANCE value instead of GRID_MAX_POINTS_X/GRID_MAX_POINTS_Y (if it's defined). This would allow it to take into account the other parameters that already work for BILINEAR such as FBLR or H, without having to code in a P parameter for BILINEAR. 

I, of course, totally agree with what the OP requested, add the P param to G29 for the sake of cross-firmware compatibility. My hope is to add to this concept in a way that then doesn't require extra G29 params (if none are desired).

Thoughts?

aalku, 2021-09-22:
I think in Bilinear GRID_MAX_POINTS_X/Y should act as max and default but optionally in G29 you should be able to request a grid with less points. The memory reserved would be for MAX points so you can`t request more than that in G29.

thisiskeithb, 2021-09-24:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/18460

closed 2021-09-24 by thisiskeithb