MarlinFirmware/Marlin #22688: [BUG] (bug summary) DELTA_AUTO_CALIBRATION requires a probe or LCD Controller while having manual probe

ricky0178, 2021-09-01:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to build the latest marlin for my custom built 3d printer. I have enabled AUTO_CALIBRATION and since i do not have a proble I have enabled PROBE MANUALLY in the probing section. When I try to compile the firmware sanitycheck gives me error:
`Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1437:6: error: #error "DELTA_AUTO_CALIBRATION requires a probe or LCD Controller."`

### Expected behavior

I expected to compile without problems, since I have enabled the manual probe option that has a comment explaining just how this manual probe is made for auto calibration

### Version of Marlin Firmware

02000901

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

OctoPrint

### Additional information & file uploads

[Configuration&adv.zip](https://github.com/MarlinFirmware/Marlin/files/7092941/Configuration.adv.zip)

Vertabreak, 2021-09-01:
the configuration does not have a lcd controller enabled with a encoder that is required to operate manual probing.

closed 2021-09-01 by thisiskeithb

ricky0178, 2021-09-01:
But as the documentation state I could use manual terminal commands to make it work, that's what made me think it was possible

ricky0178, 2021-09-02:
[image]
As expected I can auto bed level with a manual probe, as written [https://marlinfw.org/docs/features/auto_bed_leveling.html](url) in the docs here too