MarlinFirmware/Marlin #22689: [BUG] SKR 2 rev B with BTT TMC 2208 v3 -> TMC CONNECTION ERROR

VanessaE, 2021-09-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I am copying a bug report @howardjones filed in BigTreeTech's SKR-2 repo, so that I can add my remarks here so Marlin devs can see what's going on.

Here's his original post (https://github.com/bigtreetech/SKR-2/issues/43):

-----
I have a new SKR 2 rev B board and 5 BTT TMC2208 v3 stepsticks - I am getting "TMC CONNECTION ERROR" from the display at boot up, and M122 just shows:

```
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
```
Nothing else is connected currently, apart from the BTT 12864 RGB mini display, and power/serial from USB. The jumper is set underneath each socket to enable UATRT. As far as I can see, the TMC2208 modules already have a link on the UART config jumper J2 (it's really tiny, but I think it is there). Link is between the bottom two pins.

Marlin 2.0.9.1 is configured for TMC2208 on all axes.

What else is needed?

-----
My remarks:

My v1.4 started malfunctioning recent as cheap Chinese electronics are wont to do, and with none to be found anywhere for a reasonable price (no doubt due to fallout from the ongoing plague), I was left with the choice of either buying a v1.3 of uncertain quality, or buying a v2. I chose the latter, since it should have been an upgrade and it wasn't expensive.  Seemed like a good idea at the time, right?

Now the above-quoted problem is happening for me as well, using 2208's that I know for a fact are in working condition.  In fact, similar to Howard, I'm using a couple of BTT 2208 v3's (plus two older 2208's from Eryone)

There are some interesting things going on:

1.  All four of my 2208's don't necessarily report "ALL LOW" -- sometimes one or more, rarely all four, will report "OK", completely randomly from one power cycle to the next (without touching the code or the hardware).
2. Sometimes, if one or more reports "ALL LOW" after a power cycle, those same ones will report "OK" if I hit the reset button.
3. Trying to move any axis, whether it's "OK" or "ALL LOW", usually fails, giving no motion on one motor or another.  That gets a bit wonky on CoreXY's like mine - trying to home X, for example, might see the A motor move while the B motor just sits there.  Or vice-versa. 
4. It's inconsistent, but sometimes trying to move a motor whose driver reported "ALL LOW" causes that driver to get way hotter than one that reported "OK".  It's as though the one is being slammed to its maximum current.
5. If you look at the spew from `M122`...  it seems like the more drivers that return "ALL LOW" at the end, the slower the whole message is printed -- but no matter how fast or slow it prints, I do seem to get the entire message.

In fact, the overall behavior bears a strong resemlence to how things were early on in the SKR v1.1 single-wire UART work, before @gloomyandy got it going well. 

I tested my spare A4988's, and they still work fine, but... \*cringes at how loud they are* ...is there a proper solution here?

@Guilouz, you remarked that this is normal in "USB mode".  If by that you mean setting `default_envs = BIGTREE_SKR_2_USB` in `platformio.ini`, then that isn't the whole story.  This happens in plain `BIGTREE_SKR_2` mode also.  Either way, definitely not "normal".

Before writing this, I made sure all platformio-related modules/libraries/etc. were up-to-date in addition to what gets brought in automatically when compiling Marlin (on that note, TMCStepper is at v0.7.3).

### Bug Timeline

Actual age/timeline unknown, started for me today because of new hardware.

### Expected behavior

Since the v2 is marketed clearly as an upgrade over and replacement for the v1.4, things should work as similarly from one to the other as possible.  Eithewr way, the firmware should talk to the drivers properly, axes should move as commanded, etc.  You know, all the usual stuff involved in printing.

### Actual behavior

Motors don't move, drivers don't communicate properly.  Have to fallback to my old A4988's to make it work.

### Steps to Reproduce

1. Get an SKR v2 and a handful of TMC2208's
2. Set them up as you would on any other board
3. Try to move/home/etc.  It won't.

### Version of Marlin Firmware

 `bugfix-2.0.x` branch, commit 01d1192a

### Printer model

Custom-built Hypercube

### Electronics

SKR v2, no other electronics of significance

### Add-ons

No add-ons of significance.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

Pronterface

### Additional information & file uploads

[Marlin-configs-20210901.zip](https://github.com/MarlinFirmware/Marlin/files/7095151/Marlin-configs-20210901.zip)

thisiskeithb, 2021-09-02:
They need to power the board with 12/24v.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-09-02 by thisiskeithb

VanessaE, 2021-09-02:
"They" would presumably include me, and my board *is* powered by 24v.

This is not a support request, this is a bug report.  Please re-open.

VanessaE, 2021-09-02:
Actually, skip it.  I'll just re-file this without Howard's commentary.