MarlinFirmware/Marlin #22691: [BUG] malfunctioning UART comms with TMC220x drivers

VanessaE, 2021-09-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Recently, my v1.4 started malfunctioning, as cheap Chinese electronics are wont to do without warning, and with none to be found anywhere for a reasonable price due to fallout from the ongoing plague, I was left with the choice of either buying a v1.3 of uncertain quality, or buying a new SKR v2. I chose the latter, since it should have been an upgrade and it wasn't expensive.  Seemed like a good idea at the time.

With this new board, when using TMC2208 driver modules and other hardware that I know for a fact is in good, working condition, I just can't get UART going.  Physically, set-up of the v2 is all but identical to the v1.4, and in general, it's supposed to behave identically as well (of course keeping in mind that the microcontrollers are wildly different).  On the v1.4, UART Just Worked™.  On this v2, not so much.

On power-on, the display reports "TMC connection error", and the output of `M122` will show one or more drivers as "ALL LOW".  Obviously, the jumpers are properly set on the motherboard for UART mode, and Marlin's configured for it as per usual.

There are some interesting things going on:

1.  My 2208's won't necessarily \*all* report "ALL LOW" -- usually only one or two will; even all four will report "OK" rarely (but they won't actually work).  Which ones do this, varies from one power-up to the next, without any apparent pattern.
2. Sometimes, if one or more drivers report "ALL LOW" after a power cycle, those same drivers will report "OK" if I just hit the reset button.
3. Trying to home X/Y, or jog with a simple `G1` move, whether the corresponding driver is "OK" or "ALL LOW", usually just gives no motion on one motor or another.  Sometimes one motor will turn while the other stays put.
4. It's inconsistent, but sometimes trying to do a move involving a motor whose driver reported "ALL LOW", causes that driver to get way hotter than one that reported "OK".  It's as though that driver is being set to its maximum current (versus my normal setting of just 1 amp).
5. If you look at the spew from `M122`...  it seems like the more drivers that return "ALL LOW" at the end, the slower the whole message is printed -- but no matter how fast or slow it prints, I do seem to get the entire message.  It's as though Marlin is "fighting" to read all of the data from the drivers that it needs to fill-out the message.

In fact, the overall behavior bears a strong resemblence to how things were early on in the SKR v1.1 single-wire UART work, before @gloomyandy got it going well. 

I tested my spare A4988's, and they still work fine. \*cringes at how loud they are* 

Before writing this, I made sure all platformio-related modules/libraries/etc. were up-to-date in addition to what gets brought in automatically when compiling Marlin.

After a lot of combing through my configs, checking for bad hardware/wiring/etc., triple-checking jumper settings and the board's documentation, and generally googling around trying to troubleshoot this, I've concluded that it's gotta be a bug in either Marlin or the TMCStepper library (v0.7.3).

Please note:  my board is properly-powered (24v), the USB/Vdd jumper is of course properly-set, and everything is otherwise hooked-up in the same manner as it would be for actual printing (heaters, thermistors, fans, motors, end stops, etc).  Literally all that remains to be done is to bolt the motherboard's enclosure to the printer's frame and put the fans on, if I didn't need to mess with the driver modules, that is.

### Bug Timeline

Actual age/timeline unknown, started for me today because of new hardware.

### Expected behavior

Since the v2 is intended as an upgrade over and replacement for the v1.4, things should work as similarly from one to the other as possible.

However they compare, the firmware should at least talk to the drivers reliably, motors should move as commanded, etc.  You know, all the usual stuff involved in printing.

### Actual behavior

Motors don't move, drivers don't communicate properly. 

All other functions of the printer work fine -- the BLtouch self-tests on power-up, I can heat up the hotend and bed, the cold-end and motherboard fans turn on when needed, the 12864 display works, end stops trigger properly (when a driver's willing to move at ll), etc.

When I switched to my old A4988's, motors started behaving normally.  In theory, right now I could do all the usual calibration and EEPROM settings, and print something.  That's assuming I didn't mind hearing my printer singing two rooms away. :wink:

### Steps to Reproduce

1. Get an SKR v2 and a handful of TMC2208's
2. Set them up as you would on an SKR v1.3 or v1.4.
3. Make sure the board is powered from 12/24v.
4. Look at the output from `M122`, try to move an axis/home/etc.

### Version of Marlin Firmware

`bugfix-2.0.x` branch, commit 01d1192a

### Printer model

Custom-built Hypercube

### Electronics

brand new SKR v2, TMC2208's (UART-ready); no other electronics of significance

### Add-ons

No add-ons of significance.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

Pronterface

### Additional information & file uploads

[Marlin-configs-20210901.zip](https://github.com/MarlinFirmware/Marlin/files/7095415/Marlin-configs-20210901.zip)

Note: the above configuration may show A4988's in use, which would be leftover from testing with them to make sure basic motion still worked.  The only difference from the way my configs would normally be is having commented-out my TMC2208 driver selection (thus invoking the default A4988 selection), of course, and the settings to invert the motors' directions.  All other parts of the configs are unchanged.  Obviously, the motherboard's per-module shunt jumpers are set to match the driver selection.

<details><summary>Example output from `M122`, showing a driver not communicating properly...</summary>

```
SENDING:M122
		X	Y	Z	E
Enabled		false	false	false	false
Set current	1000	1000	1000	700
RMS current	994	994	994	673
MAX current	1402	1402	1402	949
Run current	17/31	17/31	17/31	21/31
Hold current	8/31	8/31	8/31	10/31
CS actual	8/31	8/31	0/31	10/31
PWM scale
vsense		0=.325	0=.325	0=.325	1=.18
stealthChop	true	true	false	false
msteps		16	16	256	16
interp		true	true	false	true
tstep		max	max	0	max
PWM thresh.	65	65	31625	52
[mm/s]		152	152	1	16
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	10	10	0	12
pwm scale auto	0	0	0	0
pwm offset auto	36	36	0	36
pwm grad auto	14	14	0	14
off time	4	4	0	4
blank time	24	24	16	24
hysteresis
 -end		2	2	-3	2
 -start		1	1	1	1
Stallguard thrs
uStep count	8	8	0	8
DRVSTATUS	X	Y	Z	E
sg_result
stst				*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C

120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:08:00:00
		Y	0xC0:08:00:00
		Z	0x00:00:00:00	 Bad response!
		E	0x80:0A:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... Error: All LOW
Testing E connection... OK
>>> m122
SENDING:M122
		X	Y	Z	E
Enabled		true	true	false	false
Set current	1000	1000	1000	700
RMS current	994	994	994	673
MAX current	1402	1402	1402	949
Run current	17/31	17/31	17/31	21/31
Hold current	8/31	8/31	8/31	10/31
CS actual	15/31	15/31	0/31	10/31
PWM scale
vsense		0=.325	0=.325	0=.325	1=.18
stealthChop	true	true	false	false
msteps		16	16	256	16
interp		true	true	false	true
tstep		max	max	0	max
PWM thresh.	65	65	31625	52
[mm/s]		152	152	1	16
OT prewarn	false	false	false	false
triggered
 OTP		false	false	false	false
pwm scale sum	40	15	0	12
pwm scale auto	0	0	0	0
pwm offset auto	93	36	0	36
pwm grad auto	14	14	0	14
off time	4	4	0	4
blank time	24	24	16	24
hysteresis
 -end		2	2	-3	2
 -start		1	1	1	1
Stallguard thrs
uStep count	168	168	0	24
DRVSTATUS	X	Y	Z	E
sg_result
stst				*
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
		X	0xC0:09:00:00
		Y	0xC0:09:00:00
		Z	0x00:00:00:00	 Bad response!
		E	0x80:0A:00:00
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... Error: All LOW
Testing E connection... OK
```
</details>

closed 2021-09-02 by VanessaE

VanessaE, 2021-09-02:
It would seem it's needed on Rev B boards as well.  Mine's a Rev B for sure (it has two QC stickers and the "correct" MOSFET).

In other words, it would seem this feature is broken at the hardware level.

thisiskeithb, 2021-09-02:
> It would seem it's needed on Rev B boards as well.

It's not. I run several SKR 2 Rev. Bs and they are running a mix of TMC2225s/22208s & TMC2226s/TMC2209s without enabling `DISABLE_DRIVER_SAFE_POWER_PROTECT`.

I would guess some other wiring/hardware issue.

VanessaE, 2021-09-02:
Well I'm at a loss to explain how else my drivers got damaged.  They were in perfect working order when I put them on.