MarlinFirmware/Marlin #22695: TMC 2130 tmc connection error wo/sdcard out and Printer HALTED Please Reset w/sdcard in

sergiofagundes, 2021-09-02:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I'm using a Delta printer with Rumba board (1st gen) TMC2130 drivers on XYZ and TMC2100 on E. 

When the SDCARD is not inserted the LCD shows: "tmc connection error".

If the sdcard is inserted the printer after a few seconds halt and show: "Driver error. Printer HALTED Please Reset"

In old Marlin versions I have not this messages or problems. I'm using the latest version of Marlin.

GCODE M122 doesn't show any problems with the drivers.

### Bug Timeline

new

### Expected behavior

no error message

### Actual behavior

Error when sdcard inserted or not varies

### Steps to Reproduce

1. No sdcard inserted
2. LCD shows "tmc connection error"
3.  sdcard inserted
4.  LCD shows: "Driver error. Printer HALTED Please Reset"

### Version of Marlin Firmware

2.0.9.1

### Printer model

Delta

### Electronics

Rumba Board 1s gen (8 bit)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

ellensp, 2021-09-02:
https://reprap.org/wiki/RUMBA ?
This board does not support UART mode, not without you making changes to the pcb

sergiofagundes, 2021-09-02:
> https://reprap.org/wiki/RUMBA ?
> This board does not support UART or SPI control, not without you making changes to the pcb or adding wire links...
> 
> As far as I can tell with the information provided you need to use TMC2130_STANDALONE

I use a wire link (cables from drivers to the rumba board).