MarlinFirmware/Marlin #22705: [BUG] Printer loses Position, while Bed Leveling

n1c030, 2021-09-03:
### Did you test the latest `bugfix-2.0.x` code?

No, but I will test it now!

### Bug Description

I use the Marlin Version of MucMoe on my A20M. And the problem is that while creating a Mesh with UBL or Bilinear at some point the printer just drives not far enough or stays in the last position. But it moves on as it is on the right position. I tried different Steppers (Original and TMC2208), changed the stepper motor. But it has to be the firmware, because when I flash on the Version 1.1.8 from Geeetech it performs the Bed Leveling fine.

### Bug Timeline

Started with upgrading the Steppers to TMC2208 and therefore using the new firmware

### Expected behavior

I expected it to create a Mesh

### Actual behavior

Drives into X-End and i have to power off the printer

### Steps to Reproduce

Start 

### Version of Marlin Firmware

2.0.7

### Printer model

Geeetech A20M

### Electronics

Geeetech GT2560 V4

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

Rubenazzzo, 2021-10-02:
me quedo, me pasa lo mismo en mi anet-a8 con inductivo y board skr 1.3

Tannoo, 2021-11-24:
You need to talk to this MucMoe guy, is this he: [https://www.thingiverse.com/mucmoe/designs](https://www.thingiverse.com/mucmoe/designs)?

We really can't support everyone else's version of Marlin.

You could ask someone at one of these places for support:

> Marlin Support
The Issue Queue is reserved for Bug Reports and Feature Requests. To get help with configuration and troubleshooting, please use the following resources:
>
>Marlin Documentation - Official Marlin documentation
Marlin Discord - Discuss issues with Marlin users and developers
Facebook Group "Marlin Firmware"
RepRap.org Marlin Forum
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube

closed 2022-02-04 by github-actions[bot]